realsenseai/librealsense #7527: I want to do a background diff

YamasouA, 2020-10-08:
Hi. 
I want to do a background diff .But I can't do that.
May be I can't understand  datatype about frame.
Please teach me!!

MartyG-RealSense, 2020-10-08:
Hi @YamasouA   Could you explain more about what you want to do please to make sure that I understand you correctly about what a **background diff** is.  Thanks!

YamasouA, 2020-10-08:
Thanks for the reply.
Sorry for my poor English.

I want to keep one template and calculate and display the current difference between it and the image.
Because Because I want to determine if people are there.
I'm currently using the openFrameworks add-on. It converts a camera frame to a cv::mat, calculates the difference between it and the current frame and then converts the image to an ofImage.
But it doesn't work.

Let me know if I haven't explained it well enough.
Thanks!

YamasouA, 2020-10-08:
[image]
This is the picture I saved and it's clearly not right.

YamasouA, 2020-10-08:
Sorry I forgot this tag @MartyG-RealSense

MartyG-RealSense, 2020-10-08:
No problem, your English is very good.   :)  I understand now.  Thank you very much for the additional information.

As you have already created a project with openframeworks, it may be best to continue with that for now and try to get it working correctly.  As a first step, could you link me please to the openframeworks plugin for RealSense that you have been using?

YamasouA, 2020-10-09:
Thanks @MartyG-RealSense !

My code upload [here](https://github.com/YamasouA/oFrealsense)
This code is a bit different now.
I really want to save only one photo, but I only want to display it, but I don't know how. I used this code to check if I could calculate the difference.
I'm not sure if this looks crazy, but I'd appreciate your advice.

MartyG-RealSense, 2020-10-09:
I did further research on your problem.  Since there is not much information about using openframeworks with librealsense, I looked at the problem just from the viewpoint of what openframeworks on its own can do.

I found an openframeworks guide that looks at how to compare two images.  To more easily find the information in the article that is relevant to your project, I recommend doing a word find operation on the browser page for the word **compare**.

https://openframeworks.cc/ofBook/chapters/image_processing_computer_vision.html

[image]

YamasouA, 2020-10-09:
Thanks for looking into it!
It's exactly what I wanted to do!
I'm going to try it, but I'm afraid I'm going to be late in answering because I can't get into school on weekends because of the COVID19.
I'll post the results for sure.
**Thank you so much!!!!**

MartyG-RealSense, 2020-10-09:
Great news - I'm pleased I could be of help!  I look forward to your test results in the coming week.   Good luck!

YamasouA, 2020-10-12:
It's been a while.
I have managed to take a depth image and calculate the difference.
Thanks to you.
Thank you so much!
[image]

MartyG-RealSense, 2020-10-12:
Great news - thanks very much for the update!

closed 2020-10-20 by MartyG-RealSense