realsenseai/librealsense #7528: librealsense v2.38.1 does not work on kernel 5.4

fiorano10, 2020-10-08:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 5.12.7 | 
| Operating System & Version |   Ubuntu 18  | 
| Kernel Version (Linux Only)    |  5.4                                       | 
| Platform                                 | PC |
| SDK Version                            |  2.38.1                    | 
| Language                            |  C++                       | 
| Segment			|  Robot                      | 

### Issue Description

I have been running librealsense v2.33 on kernel 5.4 fine but if I upgrade to v2.38.1 I keep getting the 
`Could not get Frame metadata error`

I do not want to build from source, why does 2.33 work but 2.38 does not?

MartyG-RealSense, 2020-10-08:
Hi @fiorano10   I could not find information about any error by that name.  Is it the **Frame metadata isn't available!** that you are receiving?

Also, can you confirm which method you are using to build librealsense 2.38.1 with please (source code or packages).

fiorano10, 2020-10-08:
Yup, that’s the error I get. Just installed via apt
`sudo apt install librealsense2`

MartyG-RealSense, 2020-10-08:
So you have performed a package installation with **sudo apt-get install librealsense2-dkms** then?

There are few references about metadata problems with the packages.  Of the two successful conclusions that I have seen, doing a clean reinstall of Ubuntu solved the problem in both cases.

fiorano10, 2020-10-08:
A clean install is not really an option, is there anything else that can done?

MartyG-RealSense, 2020-10-09:
If a full clean uninstall is not an option then you could try uninstalling all of the librealsense versions that you have currently got installed that were installed with the **distribution** package method.

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#uninstalling-the-packages

The above process will not remove any librealsense versions on your computer that were built from source code.

ev-mp, 2020-10-11:
@fiorano10 , the error you reported "Could not get Frame metadata error" is not in the SDK's source code and is probably generated by external wrapper/app.
I suggest to run Realsense-viewer and validate streaming, controls and metadata to establish whether it is related 
```sudo apt install librealsense2-utils=2.38.1...```

As for the patches DKMS for 5.4 is is not available, so if the kernel is not patched (which sounds unreasonable, as you mention that with LRS 2.33.1 metadata works) then you can invoke the patches scripts from the installation guide.

MartyG-RealSense, 2020-10-11:
Hi @ev-mp   Thanks very much for your help.  I think what @fiorano10  meant with that error name in the opening message is that they were actually receiving **Frame metadata isn't available!**, which they confirm with "Yup, that's the error I got" when I asked about it.  Thanks!

MartyG-RealSense, 2020-10-20:
Hi @fiorano10   Do you still require assistance with this case, please?   Thanks!

fiorano10, 2020-10-20:
Yes, no fix was suggested that worked

MartyG-RealSense, 2020-10-20:
@fiorano10   You mentioned above that a clean wipe and reinstall of Ubuntu was not really an option.   Combined with your problems with dynamic_reconfigure in https://github.com/IntelRealSense/realsense-ros/issues/1431 it suggests that there are some wider problems in your system than just the kernel, unfortunately.

MartyG-RealSense, 2020-10-30:
Hi @fiorano10  Have your problems in this case been resolved yet (bearing in mind that your computer seemed to have more problems than just the kernel).   Thanks!

closed 2020-11-07 by MartyG-RealSense