realsenseai/librealsense #7531: get_data() seems to return no data from points while get_vertices() does

collinbrake, 2020-10-08:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 2.39 | 
| Operating System & Version |   Linux, Arch  | 
| Kernel Version (Linux Only)    |   | 
| Platform                                 | PC |
| SDK Version                            |    2.39                      | 
| Language                            |  python | 
| Segment			|  Robot  | 

### Issue Description

Hi, I am am trying to convert a realsense pointcloud to a numpy array:

```code
 **frame_points = rs_ptcloud.calculate(frame_depth)                                                                                                        
                                                                                                                                                            
    # Convert the frame of points to a point cloud stored in a numpy array                                                                                  
    points = np.asanyarray(frame_points.get_vertices())                                                                                                     
    print(frame_points.get_data_size(), frame_depth.get_data_size())                                                                                        
    print(points, points.shape)
```

With `get_vertices()`, the numpy array contains data. When I substitute `get_data()`, however, the array is empty. Thanks in advance!

collinbrake, 2020-10-08:
I could be wrong, but it seems that the `get_data()` method traces back to [this definition](https://github.com/IntelRealSense/librealsense/blob/f7cdf6e8961e1709f6d864bdb33095c00a671ca7/wrappers/python/pyrs_frame.cpp#L20) for `get_frame_data()`. This function appears to return a void `BufData` type if the frame is not a video frame (`<frame>.is_video_frame == False`):

```code
if (auto vf = self.as<rs2::video_frame>()) {

...

}
else
    return BufData(const_cast<void*>(self.get_data()), 1, std::string("@B"), 0); };
```

A point frame is not a video frame, so this could be the reason that the buffer contains no data when `.get_data()` is called on a point frame.

It could also be an issue with my usage of `np.asanyarray()`, but this doesn't seem likely since this method of converting pyrealsense2 frames into numpy arrays works for color and depth frames.

MartyG-RealSense, 2020-10-08:
Hi @collinbrake   Others have explored the subject of converting frame data to Python numpy array and vice versa, and it is an ongoing area of study for RealSense users.  The link below may provide you with some research leads.

https://support.intelrealsense.com/hc/en-us/community/posts/360047819294/comments/360011874874

collinbrake, 2020-10-08:
@MartyG-RealSense Thanks for the links! I will keep digging. It looks like both of these issues are related to saving point clouds as `.ply` files and then reading them back in and trying to convert from a numpy array to a realsense frame. I am trying to do just the opposite, but without saving the point cloud to a '.ply' file first.

sam598, 2020-10-09:
@collinbrake Open3D has a pretty efficient way of converting depth maps into point clouds:
http://www.open3d.org/docs/release/tutorial/Basic/rgbd_image.html

and then getting numpy arrays from that pointcloud.
http://www.open3d.org/docs/release/tutorial/Basic/working_with_numpy.html#From-open3d.PointCloud-to-NumPy

MartyG-RealSense, 2020-10-17:
Hi @collinbrake   Do you require further assistance with this case, please?   Thanks!

collinbrake, 2020-10-17:
@MartyG-RealSense Thanks for asking - the `get_vertices()` function actually does exactly what I need. I should have experimented with it more in the first place - I was just expecting the 'get_data()` method to work like it does for other frames, and went down a rabbit trail.

collinbrake, 2020-10-17:
@sam598 Thanks for your suggestions on this!

closed 2020-10-17 by collinbrake

MartyG-RealSense, 2020-10-17:
Great news @collinbrake  - thanks very much for the update!

Yes, thanks too @sam598 for your assistance    :)