realsenseai/librealsense #7532: How to getting started with two Intel RealSense cameras and OpenCV?

ZiSo89, 2020-10-08:
| | |
|---|---|
|**librealsense**|2.38.1 RELEASE|
|**OS**|Windows|

Hi, actually there is not an issue, but I don't know where to post my question.

I am new in the field and this sound very silly for a C++ developer with experience.

So I have two cameras D430 and I would like to use OpenCV to display the depth, exactly like the "imshow" example do but with two windows one for each camera. 
I know I have to combine the "multicam" and "imshow" example but how? A simple code example will help me a lot.. :) 

If this not easy maybe an example like the "hello-realsense" but with two cameras will give me the base idea.

Thank you in advance.

sam598, 2020-10-09:
There are probably a couple of way to handle this. What I prefer to do in python is first get a list of the serial numbers of all of the attached RealSense cameras:

```
ctx = rs.context()
serials = []
for i in range(len(ctx.devices)):
    sn = ctx.devices[i].get_info(rs.camera_info.serial_number)
    print(sn)
    serials.append(sn)
```

Then I would start a unique thread for each camera. You set the serial number in the config which is passed to the pipeline:

```
config = rs.config()
config.enable_device(serials[0])

pipeline = rs.pipeline()
pipeline.start(config)
```

From there it is up to you how you want to combine the images. For a simple display you can use OpenCV's concatenate function to stack the images horizontally or vertically.

MartyG-RealSense, 2020-10-09:
@sam598   Thanks very much for your advice!

@ZiSo89  I researched the question and could not find a C++ based OpenCV example, though there was a tutorial for Python and OpenCV that adapts the **opencv_viewer_example.py** Python example program for multicam to open multiple windows, one for each camera.

https://rahulvishwakarma.wordpress.com/2019/08/17/realsense-435i-depth-rgb-multi-camera-setup-and-opencv-python-wrapper-intel-realsense-sdk-2-0-compiled-from-source-on-win10/

ZiSo89, 2020-10-09:
Actually I have one camera at the moment but I wrote the bellow script, Is it correct, will this work, or is a better structure?

`
   int main()
    {
	const auto window_name = "Display Image";
	namedWindow(window_name, WINDOW_AUTOSIZE);

	context ctx; // Create librealsense context for managing devices
	map<string, colorizer> colorizers; // Declare map from device serial number to colorizer (utility class to convert depth data RGB colorspace)
	vector<pipeline> pipelines;
	// Capture serial numbers before opening streaming
	vector<string> serials;
	for (auto&& dev : ctx.query_devices()) {
		serials.push_back(dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER));
	}

	for (int i = 0; i < serials.size(); i++) {
		printf(serials[i].c_str());
	}

	// Start a streaming pipe per each connected device
	for (auto&& serial : serials)
	{
		pipeline pipe(ctx);
		config cfg;
		cfg.enable_stream(RS2_STREAM_DEPTH, 848, 100, RS2_FORMAT_Z16, 100);
		cfg.enable_device(serial);
		pipe.start(cfg);
		pipelines.emplace_back(pipe);
		// Map from each device's serial number to a different colorizer
		colorizers[serial] = colorizer();
	}

	// We'll keep track of the last frame of each stream available to make the presentation persistent
	map<int, frame> render_frames;

	while (waitKey(1) < 0 && getWindowProperty(window_name, WND_PROP_AUTOSIZE) >= 0)
	{
		// Collect the new frames from all the connected devices
		vector<frame> new_frames;
		vector<depth_frame> depths;
		for (auto&& pipe : pipelines)
		{
			frameset fs;
			if (pipe.poll_for_frames(&fs))
			{
				for (const rs2::frame& f : fs)
				{
					new_frames.emplace_back(f);
					// Try to get a frame of a depth image
					depths.emplace_back(fs.get_depth_frame());
				}
			}
		}

		for (int i = 0; i < depths.size(); i++)
		{
			// Query the distance from the camera to the object in the center of the image
			float dist = depths[i].get_distance(848 / 2, 100 / 2);
			// Print the distance
			cout << "The camera: " << i << " is facing an object " << dist << " meters away \r";
		}

		// Convert the newly-arrived frames to render-friendly format
		for (const auto& frame : new_frames)
		{
			// Get the serial number of the current frame's device
			auto serial = rs2::sensor_from_frame(frame)->get_info(RS2_CAMERA_INFO_SERIAL_NUMBER);
			// Apply the colorizer of the matching device and store the colorized frame
			render_frames[frame.get_profile().unique_id()] = colorizers[serial].process(frame);

		}

		vector<Mat >images;

		for (const auto& r : render_frames)
		{
			// Get the depth frame's dimensions
			const int w = r.second.as<rs2::video_frame>().get_width();
			const int h = r.second.as<rs2::video_frame>().get_height();
			// Create OpenCV matrix of size (w,h) from the colorized depth data
			images.push_back(Mat(Size(w, h), CV_8UC3, (void*)r.second.get_data(), Mat::AUTO_STEP));
		}

		// Update the window with new data
		for (Mat& img : images)
		{
			imshow(window_name, img);
		}
	}
}
`

closed 2020-10-15 by ZiSo89