realsenseai/librealsense #7533: All text is displayed as squares when opening Realsense Viewer

winer1997, 2020-10-09:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR305 | 
| Firmware Version                   | 3.27.3.0 | 
| Operating System & Version |  Linux raspberrypi                 | 
| Kernel Version (Linux Only)    |   4.14.34-v7                                      | 
| Platform                                 | Raspberry Pi 3 Model B  |
| SDK Version                            |  2.0                           | 
| Language                            |  - | 
| Segment			|  Robot                          | 

### Issue Description
I use Raspberry Pi 3model B to follow [this tutorial](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_raspbian.md) step by step (including the system and software version used are the same as the tutorial), but when I open the realsense viewer, all the text Can not be displayed normally as shown on the picture. Does anyone know what went wrong?
[image]

MartyG-RealSense, 2020-10-09:
Hi @winer1997   There was another case with this problem on a Raspberry Pi Compute Module 3+ recently.  The 3+ is equivalent to a Pi 3 Model B.   

The fonts may be rendered corrupted like this if there is not enough memory for the texture atlas.  This may be corrected by making an edit to the **model-views.cpp** file and rebuilding the Viewer.

Alternatively, you could edi the configuration file located at **~/.realsense-config.json** (which does not require rebuilding the Viewer).

https://github.com/IntelRealSense/librealsense/issues/4558

clubTchief, 2020-10-17:
Installed successfully on Raspberry Pi 3B+. The realsense-viewer app works fine except for the squarish fonts - other apps appear to be unaffected. Also, modifying performance.font_oversample has no effect on the fonts displayed in the app. Please suggest for correct rendering of fonts and UI text in the app.

MartyG-RealSense, 2020-10-17:
Hi @clubTchief   A bug report has been filed with Intel already so that the issue can be investigated.  There is no further information that I can provide at this time.

clubTchief, 2020-10-25:
Thanks for being here!
Now, I switched my RPi-3B (1GB) with a new RPI-4B (8GB) and updated my Raspbian - and all the fonts came back - No change in build or install - so I believe the source is good. Problem solved for me!

MartyG-RealSense, 2020-10-25:
Thanks very much @clubTchief for the news of the successful resolution of your particular problem after a board model upgrade!

closed 2021-08-17 by ev-mp