realsenseai/librealsense #7536: I am using 3.0 USB but in intel realsense it is showing 2.0

ranjitkathiriya, 2020-10-10:
| | |
|---|---|
|**librealsense**|2.38.1 RELEASE|
|**OS**|Windows|

Please provide a description of the problem

MartyG-RealSense, 2020-10-10:
Hi @ranjitkathiriya  Could you please first try unplugging the camera, waiting a couple of seconds and then re-inserting it into the USB port with a quick and firm motion.  This should help to eliminate the possibility of whether a slow insertion into the USB port is causing the camera to be mis-detected as being on a USB2 connection.

If that does not make a difference, then please open the **RealSense Viewer** program, go to the **More** option at the top of its options side-panel and select **Hardware Reset** from the More option's menu to see whether resetting the camera whilst it is plugged into the port makes a difference to its USB status.

burowvrmagic, 2020-10-13:
I had a few days ago similar issues with 2.38.1 (using the SDK which printed an error message to log and also using the realsense viewer gui). It turned out that plugging our cam (T265) directly to the PC yielded recognition as USB3 but inserting a USB3 hub in between led to recognition as USB2. Using Windows10.

MartyG-RealSense, 2020-10-13:
Hi @burowvrmagic   I am not able to handle T265-related issues as another member of the RealSense team is responsible for those.  Could you create a new case for your issue using the **New Issue** button on the front page of the RealSense GitHub forum (linked to below) and include 'T265' in the message title, please?  Thanks!

https://github.com/IntelRealSense/librealsense/issues

closed 2020-10-20 by ranjitkathiriya