realsenseai/librealsense #7538: T265 tracking module stream causes D455 rgb camera stream to stop in Intel Realsense Viewer v2.39.0

mwaychoff, 2020-10-10:
| Camera Model                       | D455 | 
| Firmware Version                   | 05.12.08.200 | 
| Camera Model                       | T265 | 
| Firmware Version                   | 0.2.0.951 | 
| Operating System & Version | Windows10Home | 
| Platform                                 | intel NUC |
| SDK Version                            | 2.39.0.2337 | 
| Segment			|indoor mapping| 

### Issue Description
T265 tracking module stream causes D455 rgb camera stream to stop in Intel Realsense Viewer v2.39.0.

I have tried 2different D455 cameras, 3different usb3.2 cables for D455 cameras, 3different frames per second, & am using out of box 3.1 cable for T265.
With D455 rgb camera stream running as soon as you turn on T265 tracking module stream this is when you will receive error 'No Frames Received!'.
Also you have to disconnect T265 & close Intel Realsense Viewer in order to start D455 rgb camera stream again.

Please advise.

BTW: I have several L515 cameras & these do not seem to affect above.

RealSenseCustomerSupport, 2020-10-13:
----------------------------------------------

Dear mwaychoff
“**_Thank you for highlighting the [Issue/Feature]. We have moved our focus to our next generation of products and consequently, we will not be addressing this [Issue/Feature] in the T265.”_**

mwaychoff, 2020-10-13:
Important news for us to know; and perhaps you would not mind providing some additional advise:

Please clarify if you are confirming this issue exists or are you merely no longer answering questions related to T265?

Would this issue also exist running cameras from command line instead of viewer?

Under what circumstances would your team be willing to work on fixing this issue if my team was willing to pay for your time?

Is this something another team could pursue fixing using existing SDK?

Is there a replacement product T265 about to be announced that you are referring to which would be better for me to use instead?

Greatly appreciated,

matthew

RealSenseCustomerSupport, 2020-10-16:
----------------------------------------------

Dear @mwaychoff
We have D435i and D455 which feature same IMU as T265 and the stream depth image
but since there is no VPU inside D4xx, you will need to develop your own VSLAM algorithm and run the algorithm on your host

closed 2020-11-03 by RealSenseCustomerSupport