realsenseai/librealsense #7539: questions about camera settings

a-m-p-m, 2020-10-10:
Hello, 

Could you please point me to: 

1. A full list of the camera settings for various RealSense cameras that can be set via camera's API (D435i, D435, and D415 cameras are of particular interest); and 

2. Information about the camera settings that affect other camera settings (for example, I see that setting Enable Auto Exposure to 1 first and then setting Exposure to an exposure value leads to Enable Auto Exposure setting on the camera switched to 0; are there any other parameters like these ones that affect each other?). 

Thanks.

MartyG-RealSense, 2020-10-10:
1.   There is an excellent unofficial documentation website that draws its content from the official documentation and enables SDK functions to be searched by type using menus at the top of the page.

https://unanancyowen.github.io/librealsense2_apireference/classes.html

It should be noted though that **Advanced Mode** functions are not fully documented.  This is because they interact with each other in complex ways, so Intel controls them with machine-learning algorithms instead of providing documentation.  RealSense users are free to do trial and error experimentation with Advanced Mode functions to see how changes affect the image though.

2.  Other than auto-exposure deactivating when exposure or gain is manually changed, the other function that comes to mind that is affected by value changes in an option is that if the **Laser Power** option is set to '0' then the laser turns off.

https://github.com/IntelRealSense/librealsense/wiki/API-How-To#controlling-the-laser

a-m-p-m, 2020-10-11:
Thank you, @MartyG-RealSense

MartyG-RealSense, 2020-10-30:
Case closed due to no fiurther comments received.

closed 2020-10-30 by MartyG-RealSense