realsenseai/librealsense #7540: module 'pyrealsense2' has no attribute 'pipeline'

neilyoung, 2020-10-10:
I apologize, I wasn't active here for a long time. I tested my code (Python, https://github.com/neilyoung/t265-realsense-poc-application) against the latest 2.39.0. I found some issues, for instance "context" and "pipeline" seem to be no longer attributes of pyrealsense2.

I stepped back to 2.36.0 and found everything OK as with the version 2.34.0, I tested with.

Would you mind to point me to the change here? When did it happen? How to deal with it?

```
>>> import pyrealsense2 as rs
>>> rs.pipeline
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: module 'pyrealsense2' has no attribute 'pipeline'
>>> rs.context
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: module 'pyrealsense2' has no attribute 'context'
>>> 

```

neilyoung, 2020-10-10:
Here some more information:

My OS is MacOS Catalina 10.15.7.

With 2.36.0 I get this:

```
Python 3.8.5 (default, Jul 21 2020, 10:48:26) 
[Clang 11.0.3 (clang-1103.0.32.62)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyrealsense2 as rs
>>> rs.pipeline()
<pyrealsense2.pipeline object at 0x1044ec4b0>
>>> rs.context()
<pyrealsense2.context object at 0x1044c35f0>
>>> 
```
With 2.39.0 this:

```
Python 3.8.5 (default, Jul 21 2020, 10:48:26) 
[Clang 11.0.3 (clang-1103.0.32.62)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyrealsense2 as rs
>>> rs.pipeline()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: module 'pyrealsense2' has no attribute 'pipeline'
>>> rs.context()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: module 'pyrealsense2' has no attribute 'context'
>>> 
```

neilyoung, 2020-10-10:
Does also not work with 2.38. Has maybe to do with a changed install directory for the libs...

2.38/39 is installing to /usr/local/lib/python3.8/site-packages/pyrealsense2/
2.36 is installing to /usr/local/lib

neilyoung, 2020-10-11:
The change happened between 2.36.0 and 2.37.0

neilyoung, 2020-10-11:
Looks like #6820

neilyoung, 2020-10-11:
... and definitely the "fix" proposed in #6820 does it:

- Replace wrappers/python/CMakeList.txt with an older version, here 2.34.0

I don't have a good feeling with this fix, because there are a couple of changes between these two files, not just one line... Anyway, after reverting the CMakeList.txt to the 2.34.0 version everything is fine again...

The question now is: How can such a problem be unknown since month? In fact since 2.36.0 Python is dysfunctional...

neilyoung, 2020-10-11:
Alternative approach: Use the right CMakeList.txt and change the PYTHONPATH. For instance:

Path in older versions:

```
export PYTHONPATH=$PYTHONPATH:/usr/local/lib
```

Path from 2.37.0 on (for instance):

```
export PYTHONPATH=$PYTHONPATH:/usr/local/Cellar/python@3.8/3.8.5/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyrealsense2
```

lancetee, 2020-10-14:
Hey nei, I have came across an issue exactly as you did, under SDK2.38,python_version=3.8, but camera works from the source code. Besides that, I use conda to create an env,python==3.6(it only support 3.5-3.7), install it from PYPI, it can also work, see if you would have a try.

neilyoung, 2020-10-14:
@lancetee In the end it was just a changed install directory for the libs. I reported a solution here https://github.com/IntelRealSense/librealsense/issues/7540#issuecomment-706737724.

For me that works. The PYPI is not a solution for me, because there is no package for macOS.

closed 2020-10-14 by neilyoung

MinaGabriel, 2020-10-15:
For me, I hade to do 

`import pyrealsense2.pyrealsense2 as rs`

TDHster, 2020-10-21:
sudo make install have a wrong place for libs. Current it in dist-packages/pyrealsense2 folder, but python search in dist-packages.
Finded workaround: 
/usr/lib/python3/dist-packages/pyrealsense2$ sudo cp pybackend2* ..
/usr/lib/python3/dist-packages/pyrealsense2$ sudo cp pyrealsense2* ..

For developers - I stronlgy ask to fix this.

PatrickNa, 2020-11-24:
> For me, I hade to do
> 
> `import pyrealsense2.pyrealsense2 as rs`

I have build pyrealsense version 2.40 with Python 3.8 on a Jetson Xavier AGX and also ran into the `module 'pyrealsense2' has no attribute 'pipeline'` issue. The quoted hint from @MinaGabriel works for me, too. However I would like it to behave as any other module with a 'simple' import. 

The following shows where everything was installed. Since I created a virtual environment it is also copying the files to the respective directory, too. 
```console
Install the project...
-- Install configuration: ""
-- Installing: /usr/local/lib/librealsense2.so.2.40.0
-- Installing: /usr/local/lib/librealsense2.so.2.40
-- Installing: /usr/local/lib/librealsense2.so
-- Up-to-date: /usr/local/include/librealsense2
-- Up-to-date: /usr/local/include/librealsense2/hpp
-- Up-to-date: /usr/local/include/librealsense2/hpp/rs_options.hpp
-- Up-to-date: /usr/local/include/librealsense2/hpp/rs_context.hpp
-- Up-to-date: /usr/local/include/librealsense2/hpp/rs_serializable_device.hpp
-- Up-to-date: /usr/local/include/librealsense2/hpp/rs_pipeline.hpp
-- Up-to-date: /usr/local/include/librealsense2/hpp/rs_internal.hpp
-- Up-to-date: /usr/local/include/librealsense2/hpp/rs_types.hpp
-- Up-to-date: /usr/local/include/librealsense2/hpp/rs_sensor.hpp
-- Up-to-date: /usr/local/include/librealsense2/hpp/rs_record_playback.hpp
-- Up-to-date: /usr/local/include/librealsense2/hpp/rs_device.hpp
-- Up-to-date: /usr/local/include/librealsense2/hpp/rs_frame.hpp
-- Up-to-date: /usr/local/include/librealsense2/hpp/rs_processing.hpp
-- Up-to-date: /usr/local/include/librealsense2/hpp/rs_export.hpp
-- Up-to-date: /usr/local/include/librealsense2/rs_advanced_mode.hpp
-- Up-to-date: /usr/local/include/librealsense2/rs.hpp
-- Up-to-date: /usr/local/include/librealsense2/rsutil.h
-- Up-to-date: /usr/local/include/librealsense2/h
-- Up-to-date: /usr/local/include/librealsense2/h/rs_record_playback.h
-- Up-to-date: /usr/local/include/librealsense2/h/rs_types.h
-- Up-to-date: /usr/local/include/librealsense2/h/rs_processing.h
-- Up-to-date: /usr/local/include/librealsense2/h/rs_config.h
-- Up-to-date: /usr/local/include/librealsense2/h/rs_sensor.h
-- Up-to-date: /usr/local/include/librealsense2/h/rs_advanced_mode_command.h
-- Up-to-date: /usr/local/include/librealsense2/h/rs_internal.h
-- Up-to-date: /usr/local/include/librealsense2/h/rs_device.h
-- Up-to-date: /usr/local/include/librealsense2/h/rs_option.h
-- Up-to-date: /usr/local/include/librealsense2/h/rs_frame.h
-- Up-to-date: /usr/local/include/librealsense2/h/rs_context.h
-- Up-to-date: /usr/local/include/librealsense2/h/rs_pipeline.h
-- Up-to-date: /usr/local/include/librealsense2/rs.h
-- Up-to-date: /usr/local/include/librealsense2/rs_advanced_mode.h
-- Installing: /usr/local/lib/cmake/realsense2/realsense2Targets.cmake
-- Installing: /usr/local/lib/cmake/realsense2/realsense2Targets-noconfig.cmake
-- Installing: /usr/local/lib/cmake/realsense2/realsense2Config.cmake
-- Installing: /usr/local/lib/cmake/realsense2/realsense2ConfigVersion.cmake
-- Installing: /usr/local/lib/pkgconfig/realsense2.pc
-- Installing: /usr/local/lib/librealsense-file.a
-- Installing: /home/xavier/venv-python38/road-detection/lib/python3.8/site-packages/pyrealsense2/pybackend2.cpython-38-aarch64-linux-gnu.so.2.40.0
-- Installing: /home/xavier/venv-python38/road-detection/lib/python3.8/site-packages/pyrealsense2/pybackend2.cpython-38-aarch64-linux-gnu.so.2
-- Installing: /home/xavier/venv-python38/road-detection/lib/python3.8/site-packages/pyrealsense2/pybackend2.cpython-38-aarch64-linux-gnu.so
-- Installing: /home/xavier/venv-python38/road-detection/lib/python3.8/site-packages/pyrealsense2/pyrealsense2.cpython-38-aarch64-linux-gnu.so.2.40.0
-- Installing: /home/xavier/venv-python38/road-detection/lib/python3.8/site-packages/pyrealsense2/pyrealsense2.cpython-38-aarch64-linux-gnu.so.2.40
-- Set runtime path of "/home/xavier/venv-python38/road-detection/lib/python3.8/site-packages/pyrealsense2/pyrealsense2.cpython-38-aarch64-linux-gnu.so.2.40.0" to ""
-- Installing: /home/xavier/venv-python38/road-detection/lib/python3.8/site-packages/pyrealsense2/pyrealsense2.cpython-38-aarch64-linux-gnu.so
-- Installing: /usr/local/lib/cmake/pyrealsense2/pyrealsense2Targets.cmake
-- Installing: /usr/local/lib/cmake/pyrealsense2/pyrealsense2Targets-noconfig.cmake
-- Installing: /usr/local/lib/cmake/pyrealsense2/pyrealsense2Config.cmake
-- Installing: /usr/local/lib/cmake/pyrealsense2/pyrealsense2ConfigVersion.cmake
-- Installing: /usr/local/lib/librealsense2-gl.so.2.40.0
-- Installing: /usr/local/lib/librealsense2-gl.so.2.40
-- Set runtime path of "/usr/local/lib/librealsense2-gl.so.2.40.0" to ""
-- Installing: /usr/local/lib/librealsense2-gl.so
-- Up-to-date: /usr/local/include/librealsense2-gl/rs_processing_gl.h
-- Up-to-date: /usr/local/include/librealsense2-gl/rs_processing_gl.hpp
-- Installing: /usr/local/lib/cmake/realsense2-gl/realsense2-glTargets.cmake
-- Installing: /usr/local/lib/cmake/realsense2-gl/realsense2-glTargets-noconfig.cmake
[... 56 lines cut ...]
-- Set runtime path of "/usr/local/bin/rs-convert" to ""
-- Installing: /usr/local/bin/rs-enumerate-devices
-- Set runtime path of "/usr/local/bin/rs-enumerate-devices" to ""
-- Installing: /usr/local/bin/rs-fw-logger
-- Set runtime path of "/usr/local/bin/rs-fw-logger" to ""
-- Installing: /usr/local/bin/rs-terminal
-- Set runtime path of "/usr/local/bin/rs-terminal" to ""
[... cut ...]
-- Installing: /usr/local/bin/realsense-viewer
-- Set runtime path of "/usr/local/bin/realsense-viewer" to ""
-- Installing: /usr/local/bin/rs-depth-quality
-- Set runtime path of "/usr/local/bin/rs-depth-quality" to ""
-- Installing: /usr/local/bin/rs-rosbag-inspector
-- Set runtime path of "/usr/local/bin/rs-rosbag-inspector" to ""
-- Installing: /usr/local/bin/rs-benchmark
-- Set runtime path of "/usr/local/bin/rs-benchmark" to ""
-- Installing: /usr/local/lib/libfw.a
```

I keep you updated once I figure it out.

PatrickNa, 2020-11-24:
In my case also the following is an option:
```console
export PYTHONPATH=$PYTHONPATH:/home/xavier/venv-python38/road-detection/lib/python3.8/site-packages/pyrealsense2/
```
Please note that I have installed it within a virtual environment.

brunovollmer, 2020-12-04:
Same problem here.

ranushw, 2021-04-21:
> For me, I hade to do
> 
> `import pyrealsense2.pyrealsense2 as rs`

This worked for me. I am using RPi3+ with Realsense d435i. Thanks.

07hokage, 2021-06-11:
> For me, I hade to do
> 
> `import pyrealsense2.pyrealsense2 as rs`

This worked for me . But i worked on it a bit on why we had to do this. Like @TDHster  mentioned, python looks  in dist-packages/pyrealsense2.  this pyrealsense2 folder has the .so objects. But to be able to treat this as a library package,  copy the __init__.py file from librealsense wrapper. 

`sudo cp librealsense/wrappers/python/pyrealsense2/__init__.py     /usr/local/lib/python3.6/dist-packages/pyrealsense/` . After doing this one can directly do `import pyrealsense2 as rs`

**Note:** I had pyrealsense2 folder present at `/usr/local/lib/python3.6/dist-packages` after its been built

ktsujister, 2021-07-26:
@07hokage 's solution worked for me.
It would be great if `__init__.py` file also gets copied to the target directory when `make install` is executed.

jash-maester, 2022-01-16:
> For me, I hade to do
>```import pyrealsense2.pyrealsense2 as rs```

Worked for me too, as of this date. Same issue with the dist-packages/pyrealsense2/*.so

For those who are having difficulty in finding the location of their pyrealsense2*.so, here's a simple shell one-liner
```find / -name "*pyrealsense2*"``` 
Few locaitons are bound to be inaccessible if executed without sudo, and that's completely fine,

imChenx, 2023-03-04:
I encountered the same situation. 
However,  I have a problem that I do not have the file `__init__.py` in `librealsense/wrappers/python/pyrealsense2/` 
So,  what should I do?

> > For me, I hade to do
> > `import pyrealsense2.pyrealsense2 as rs`
> 
> This worked for me . But i worked on it a bit on why we had to do this. Like @TDHster mentioned, python looks in dist-packages/pyrealsense2. this pyrealsense2 folder has the .so objects. But to be able to treat this as a library package, copy the **init**.py file from librealsense wrapper.
> 
> `sudo cp librealsense/wrappers/python/pyrealsense2/__init__.py /usr/local/lib/python3.6/dist-packages/pyrealsense/` . After doing this one can directly do `import pyrealsense2 as rs`
> 
> **Note:** I had pyrealsense2 folder present at after its been built`/usr/local/lib/python3.6/dist-packages`