realsenseai/librealsense #7541: There are noises in depth map after align

wangsff, 2020-10-11:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.04.01.02 | 
| Operating System & Version |  Win10 |         
| Platform                                 | PC |
| SDK Version                            | intel realsemse viewer v2.38.1}                          | 
| Language                            | python }                          | 

### Issue Description
I get RGB map and Depth map with python, but after align process, the align depth map show some noise that doesnt appear in orign depth map.
this is RGB map
[image]
this Depth map before align and you can see there is no noise on the toy
[image]
but after align, the noises appear in the depth map
[image]

Here is my codes
`""" We are planning to use the following packages,
    see that all the import load correctly"""
import cv2                                # state of the art computer vision algorithms library
import numpy as np                        # fundamental package for scientific computing
import matplotlib.pyplot as plt           # 2D plotting library producing publication quality figures
import pyrealsense2 as rs                 # Intel RealSense cross-platform open-source API
print("Environment Ready")

"""we open a stream using a record file (.bag)"""
# Setup: open stream
pipe = rs.pipeline()
cfg = rs.config()
cfg.enable_device_from_file(r"C:\Users\DELL\Documents\hhh.bag")
profile = pipe.start(cfg)

# Skip 5 first frames to give the Auto-Exposure time to adjust
for x in range(5):
    pipe.wait_for_frames()

# Store this frameset for later processing:
frameset = pipe.wait_for_frames()
color_frame = frameset.get_color_frame()
depth_frame = frameset.get_depth_frame()

# Cleanup: close stream
pipe.stop()
print("Frames Captured")

""" plot the color image """
color = np.asanyarray(color_frame.get_data())
# color = color[:,:,0] # for one of the color in the image
plt.rcParams["axes.grid"] = False
plt.rcParams['figure.figsize'] = [12, 6]
plt.imshow(color)
plt.show()

""" now plot the depth frame """
colorizer = rs.colorizer()
colorized_depth = np.asanyarray(colorizer.colorize(depth_frame).get_data())
plt.imshow(colorized_depth)
plt.show()

""" Stream Alignment
Upon closer inspection you can notice that the two frames are not captured from the same physical viewport.
To combine them into a single RGBD image, let's align depth data to color viewport: """
# Create alignment primitive with color as its target stream:
align = rs.align(rs.stream.color)
frameset = align.process(frameset)

# Update color and depth frames:
aligned_depth_frame = frameset.get_depth_frame()
colorized_depth = np.asanyarray(colorizer.colorize(aligned_depth_frame).get_data())

# Show the two frames together:
images = np.hstack((color, colorized_depth))
plt.imshow(images)
plt.show()
`
I set the below option in reviewer 2.38.1
[image]
[image]
[image]
[image]

Looking forward to your reply
Thanks!

wangsff, 2020-10-11:
Is the noise caused by upsampling? 
If so, can I resize the color frame size(1280x710) to depth frame size(1024x768)?

wangsff, 2020-10-11:
I see the color&depth show on https://www.intelrealsense.com/lidar-camera-l515/, and the color depth map align very well.
How does that work?
[image]

wangsff, 2020-10-11:
@seebq

RealSenseSupport, 2020-10-30:
Hi,

At first glance this seems like an artifact of sampling and/or the occlusion phenomenon described here:

https://dev.intelrealsense.com/docs/projection-texture-mapping-and-occlusion-with-intel-realsense-depth-cameras

If you run our python example as-is in your environment do you experience the same artifacts?

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/align-depth2color.py

Thanks

closed 2020-11-24 by RealSenseSupport