realsenseai/librealsense #7544: I am getting this error

ranjitkathiriya, 2020-10-11:
| | |
|---|---|
|**librealsense**|2.38.1 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
GLFW Driver Error: WGL: The driver does not appear to support OpenGL
Could not initialize offscreen context!

```

MartyG-RealSense, 2020-10-11:
Hi @ranjitkathiriya   I hope that the link below will be helpful.

https://github.com/IntelRealSense/librealsense/issues/6976

ranjitkathiriya, 2020-10-11:
Hi @MartyG-RealSense ,

Thanks for replying I  have solved this issue by this link once again thanks.

[image]

Then after I am getting this error when I try to open real sense viewer.

MartyG-RealSense, 2020-10-12:
Did you install the full 793 mb 2.38.1 SDK from the 'Releases' page please, and then launch the Viewer by double-leftclicking on the launch shortcut placed on the Windows desktop, please?

[image]

If you did install the full SDK, it may be worth checking the **Downloads** folder of your Windows user account to see if the file **Intel.RealSense.SDK-WIN10-2.38.1.2223.exe** is there.  The typical path to this file would be:

**C: > Users > [your user folder name] > Downloads**

You could then double-leftclick on the installer file without having to download it again and re-install SDK 2.38.1.  It should overwrite your preious installation of the SDK.

ranjitkathiriya, 2020-10-12:
Yes! I am on windows 10 and I have also tried to reinstall the application then also it is not working in my case. I have also tried different versions like Intel.RealSense.SDK-WIN10-2.38.1.2223.exe,Intel.RealSense.SDK-WIN10-2.39.0.exe Then also I am facing the same issue, but if I am trying only Intel.RealSense.Viewer.exe them I am getting two error this is attached below.

[image]

MartyG-RealSense, 2020-10-12:
Thank you very much for the information.  As the next step, can you confirm please whether the Windows privacy settings are set to permit access to the camera by applications using the instructions in the link below:

https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789

ranjitkathiriya, 2020-10-12:
[image]

Yes! I have checked that as well.

MartyG-RealSense, 2020-10-13:
Can you try to run another RealSense program please to see whether the application opening error is limited only to the Viewer.

If you right-click on the Viewer's launch icon on your Windows desktop and select the **Open file location** option then the **Tools** SDK folder containing pre-built executable versions of RealSense example programs will open.

Please double-leftclick on the **rs-capture** example program to run it and let me know what happens, please.   Thanks!

[image]

MartyG-RealSense, 2020-10-20:
Hi @ranjitkathiriya   Do you still require assistance with this case, please?   Thanks!

ranjitkathiriya, 2020-10-20:
No Thanks solved

closed 2020-10-20 by ranjitkathiriya