realsenseai/librealsense #7546: Number of points in cloud high when object near to left camera

vigneshsara, 2020-10-12:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D415 } |  
| Operating System & Version |   { Linux (Ubuntu 18)  |  
| Platform                                 | PC |                       | 
| Language                            |  {C ++}     | 

### Issue Description
Number of points in point cloud is high when object is near to left camera compare with same object away from left camera,
i did't know why this behavior happen,is this stereo behavior ??? 

camera to object distance = 700mm 
No.of point in object near to left camera =329613
No.of point in object away to left camera= 323034
[cloud_data.zip](https://github.com/IntelRealSense/librealsense/files/5363293/cloud_data.zip)

MartyG-RealSense, 2020-10-12:
Hi @vigneshsara   May I first clarify first your meaning when you say that the object is near and away from the left camera.   As you have only quoted one distance (700mm), this suggests that the object is always the same distance from the camera.  

So do you mean please that there are more points when the object is closer to the left camera ('near') and less points when the object is closer to the right camera ('away')?

vigneshsara, 2020-10-12:
HI @MartyG-RealSense 
thank for your replay,
1. object is is always the same distance from the camera.
2. "near" means object is closer to the left camera  ,"away"  means  object is closer to the right camera , and when i move the object gradually from right camera to left camera, point in the cloud also gradually increases.

MartyG-RealSense, 2020-10-12:
I have not personally experienced this issue.  I would speculate though that it is related to how the 400 Series stereo depth cameras generate depth images.

RealSense stereo cameras tend to rely on the left imager to generate the depth image.  The 400 Series cameras' technical data sheet document says: "The left and right imagers capture the scene and sends imager data to the depth imaging processor, which calculates depth values for each pixel in the image by correlating points on the left image to the right image and via shift between a point on the Left image and the Right image".

The dynamic calibration guide for the 400 Series, meanwhile, says: "Assume left camera is the reference camera and is located at world origin".

closed 2020-10-30 by MartyG-RealSense