realsenseai/librealsense #7547: Problem reading rosbag creating using command line tool recorder

DoctorMartins9, 2020-10-12:
### RS-RECORDER BUG?
We tried to customize [rs-recorder](https://github.com/IntelRealSense/librealsense/tree/master/tools/recorder) example by setting resolution and framerate manually throught "enable_stream" method. 
Here is the simple code lines that differs from the pre-made example:
```
[...]

rs2::pipeline pipe;
rs2::config cfg;
cfg.enable_record_to_file(out_file.getValue());

# Here are the modified lines from example
cfg.enable_stream(RS2_STREAM_COLOR, 640, 480, RS2_FORMAT_RGB8, 60);
cfg.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 60);

[...]
```
[rs-record.zip](https://github.com/IntelRealSense/librealsense/files/5363650/rs-record.zip)

### READ ROSBAG USING PYTHON
All works fine, but when we try to use the .bag file we catch always the following error (only when we set the framerate of the RGB sensor from 30 to 60):
```
Traceback (most recent call last):
  File "<path>", line 171, in main
    frames = pipeline.wait_for_frames()
RuntimeError: Frame didn't arrive within 5000
```
The code for reading the rosbag is the following:
```
pipeline = rs.pipeline()
config = rs.config()
rs.config.enable_device_from_file(config, "<filename>")        
config.enable_stream(rs.stream.depth, 640, 480, rs.format.any, 60)
config.enable_stream(rs.stream.color, 640, 480, rs.format.rgb8, 60)
        
# Start streaming from file
profile = pipeline.start(config)
device   = profile.get_device()
        
playback = rs.playback(device)
playback.set_real_time(False)
align = rs.align(rs.stream.color)

# Get frameset of depth
print("Wait for frame")
frames = pipeline.wait_for_frames()
print("Got frame")
```

This code works fine when reading rosbag file created by realsense-viewer gui, but not with rosbag recorded using our custom rs-recorder. The rosbag created using our custom command line recorder can be read correctly throught other ways like Matlab or the rosbag viewer embedded into realsense-viewer, so we think that the problem is in the python above.

### Requested informations
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435
| Firmware Version                   | 05.12.08.200| 
| Operating System & Version | Ubuntu 18.04  | 
| Platform                                 | PC  |
| SDK Version                            |  2.36.0                          | 
| Language                            |  C++ / Python                         |

MartyG-RealSense, 2020-10-12:
Hi @DoctorMartins9   Problems occurring with bag playback when set_real_time is false instead of true is a well known phenomenon.

The link below provides an example of a workaround in Python to avoid **Frame didn't arrived within 5000** , by setting up a profile and setting a playback time.

https://stackoverflow.com/questions/58482414/frame-didnt-arrived-within-5000-while-reading-bag-file-pyrealsense2

DoctorMartins9, 2020-10-12:
Ok @MartyG-RealSense thanks for the reply, but there isn't any "clean" method for solve this issue? This seems more like a workaround for the problem.

MartyG-RealSense, 2020-10-12:
The solution may be correct for some people and not for others.  The problem is that 'frames didn't arrived' errors can occur unpredictably and are difficult to resolve.  Some users may experience them and some may not even when running the same script with the same camera model.  So there is the possibility that any solution suggested that worked for another user may still experience 'frames didn't arrived' without a clear reason for why it occurs.

My understanding with SDK bag reading code though is that the FPS that is defined in the stream configuration (e.g 60) should be the same FPS that the bag was recorded at.  So if the bag was recorded at 30 FPS and you are setting the playback to 60 FPS, I would recommend re-recording the bag at 60 FPS and see if that makes a difference.

MartyG-RealSense, 2020-10-20:
Hi @DoctorMartins9   Do you require further assistance with this case, please?  Thanks!

DoctorMartins9, 2020-10-20:
@MartyG-RealSense if for you is ok to use workaround in an Intel's library of this dimension.....

MartyG-RealSense, 2020-10-20:
Sometimes a workaround is the only option available, unfortunately.

Would it be posible to update to SDK **2.39.0** and see if it improves your results.  There have been a few recent cases where **frames didn't arrived within** errors were resolved by updating to the latest firmware and SDK.  You already have the latest firmware at the time of writing this (**5.12.8.200**)

DoctorMartins9, 2020-10-20:
Ok, thanks for the help, I hope there will be time for fixing this!

Have a nice day

closed 2020-10-20 by DoctorMartins9