realsenseai/librealsense #7549: Issue : I can't launch rs-pcl exemples

Eleart, 2020-10-12:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 05.12.07.100 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC  |
| SDK Version                            | 2.38.1                   | 
| Language                            |  C++                     | 
| Segment			|  AR pcl                        | 

### Issue Description
 - I downloaded  and installed the lastest release of [pcl all in one](https://github.com/PointCloudLibrary/pcl/releases/tag/pcl-1.11.1)
 - I used CMake to generate visual studio 2019 project for the rs-pcl exemples
 - When the project is generate I build `rs-pcl` or `rs-pcl-color` and I have this issues : 
 `Cannot open include file: 'GLFW/glfw3.h': No such file or directory`

 - if in the include I specified an absolute path like that : ` #include <C:/GLFW/glfw-3.3.2.bin.WIN32/include/GLFW/glfw3.h>`
I have this error : cannot open input file 'glfw3.lib'

### More information
 - I already tested with x32 and x64 version of glfw 
 - When I generated the project with Cmake I saw : `GLFW 3.3 not found; using internal version`

MartyG-RealSense, 2020-10-12:
Hi @Eleart   I do not think it is related to PCL, as the error **Cannot open include file: 'GLFW/glfw3.h': No such file or directory** can occur when building other librelsense programs.   

The case in the link below has some similarities to yours in terms of the errors generated, and suggests Visual Studio linkages to add to rectify the errors:

https://github.com/IntelRealSense/librealsense/issues/4278

Eleart, 2020-10-12:
thanks for the response,

Effectively, I added the glfw3.lib path in the properties of the exemple and it works

Thank you

closed 2020-10-12 by Eleart

MartyG-RealSense, 2020-10-12:
Great to hear - thanks very much @Eleart for the update!