realsenseai/librealsense #7552: L515 doesn't turn on or respond at all.

Chandler, 2020-10-12:
Looking for any advanced debugging tips, I have a new, never used L515 which is not responsive at all when I plug it into a computer.

This is a computer/usb cable which works fine with d435/d415/d455 cameras. I tried 'rs-enumerate-devices' and 'realsense-viewer' and there is no sign of it.

Is it possible I just have a broken unit? do I need a special driver?

dorodnic, 2020-10-19:
Hi @Chandler 
Please check in device manager or dmesg that the camera was enumerated as USB3. Operating L515 in USB2 mode requires firmware update, and will not work out of the box.

Chandler, 2020-10-21:
@dorodnic interesting I found this is in dmseg

```
[37797.666619] usb 3-1: new high-speed USB device number 2 using xhci_hcd
[37797.816027] usb 3-1: New USB device found, idVendor=8086, idProduct=0b64, bcdDevice=10.41
[37797.816032] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[37797.816035] usb 3-1: Product: Intel(R) RealSense(TM) 515
[37797.816039] usb 3-1: Manufacturer: Intel(R) RealSense(TM) Camera
[37797.816041] usb 3-1: SerialNumber: 00000000F0221207
```

and yet
```
rs-enumerate-devices
No device detected. Is it plugged in?
```

Chandler, 2020-10-21:
hmm I guess the above post it's still usb2, if I use a different cable/port I get "SuperSpeed" usb which I'm told is usb3

```
[39324.759002] usb 2-2: USB disconnect, device number 22
[39336.979075] usb 2-2: new SuperSpeed Gen 1 USB device number 23 using xhci_hcd
[39337.000401] usb 2-2: New USB device found, idVendor=8086, idProduct=0b64, bcdDevice=10.41
[39337.000406] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[39337.000409] usb 2-2: Product: Intel(R) RealSense(TM) 515
[39337.000412] usb 2-2: Manufacturer: Intel(R) RealSense(TM) Camera
[39337.000415] usb 2-2: SerialNumber: 00000000F0221207
```

and device is still not recognized

Chandler, 2020-10-21:
It turns out I just needed to upgrade librealsense, thanks.

`sudo apt-get install librealsense2-utils librealsense2-dkms librealsense2-udev-rules librealsense2 `

closed 2020-10-21 by Chandler