realsenseai/librealsense #7553: Using context and sensor in user class hangs on destruction

burowvrmagic, 2020-10-13:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.2.0.951 | 
| Operating System & Version |  Win 10  | 
| Kernel Version (Linux Only)    | --                                         | 
| Platform                                 | PC  |
| SDK Version                            | 2.38.1 }                          | 
| Language                            |  C++ (VS2017, MSVC toolset vs141, Windows SDK 10.0.17763)                          | 
| Segment			| VR                          | 

### Issue Description
I have built a minimal project consisting of a main program file and two classes. It simply streams gyroscope data and
can detect if the device is disconnected and reconnected and can restart streaming. See attached text files for the program:

[IntelRealSenseControl_cpp.txt](https://github.com/IntelRealSense/librealsense/files/5369721/IntelRealSenseControl_cpp.txt)
[IntelRealSenseControl_h.txt](https://github.com/IntelRealSense/librealsense/files/5369722/IntelRealSenseControl_h.txt)
[IntelRealSenseSensor_cpp.txt](https://github.com/IntelRealSense/librealsense/files/5369723/IntelRealSenseSensor_cpp.txt)
[IntelRealSenseSensor_h.txt](https://github.com/IntelRealSense/librealsense/files/5369724/IntelRealSenseSensor_h.txt)
[main_cpp.txt](https://github.com/IntelRealSense/librealsense/files/5369726/main_cpp.txt)

SDK warnings and errors are enabled and print to output.

The program stops if you type 'c' and press enter. It happens quite often (typically at least once in 20 trials) that the program hangs during destruction in the mp_sensorManager.reset() step of the IntelRealSenseControl::controlLoop().

The pointer mp_sensorManager holds a class with a context and a sensor member.
Thus, the reset destroys these SDK objects. First sensor, then context.

### Further Details

It seems to happen in the destruction of the rs2_delete_sensor of rs.cpp.
There, VALIDATE_NOT_NULL(device) evaluates sometimes to false and a throw is issued
which leads to skipping the call of the context destructor.

However, the context destructor would stop the thread, where the device watcher is running (_device_watcher->stop()). This thread was started when the program was handed over a callback function to the API member function of context: set_devices_changed_callback.

Thus, this thread is never stopped and the controlLoop thread cannot be joined back and the program hangs.

Does this program use the API in a wrong way or is it indeed an API bug?
Thanks for help.
Asbjoern

burowvrmagic, 2020-10-13:
Just as an update: there is a potential bug in my IntelRealSenseControl.cpp source, thus I update it here:

[IntelRealSenseControl_cpp.txt](https://github.com/IntelRealSense/librealsense/files/5371355/IntelRealSenseControl_cpp.txt)

In createCallBack(), gyro_data is now defined within the functor.

This DOES NOT change the issue described above.

RealSenseSupport, 2021-05-04:
Fixed in latest via PR https://github.com/IntelRealSense/librealsense/pull/8561

Thanks

closed 2021-05-04 by RealSenseSupport