realsenseai/librealsense #7556: Problem depth align to color map. Need help!

wangsff, 2020-10-13:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.04.01.02 | 
| Operating System & Version |  Win10 |         
| Platform                                 | PC |
| SDK Version                            | intel realsemse viewer v2.38.1}                          | 
| Language                            | python }                          | 

### Issue Description
when I align depth frame to color frame with code that you provide.  Something very strange has come up. As you can see, there are some "duplication" region in aligned color map.

wangsff, 2020-10-13:
This is what the toy looks like
[image]

But after align to depth map, some strange duplication region come up

[image]

wangsff, 2020-10-15:
[image]
[code](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/align-depth2color.py)
I change `align_to = rs.stream.depth` to `align_to = rs.stream.color`

dorodnic, 2020-10-18:
I believe its effect of occlusion - https://dev.intelrealsense.com/docs/projection-texture-mapping-and-occlusion-with-intel-realsense-depth-cameras#section-4-occlusion-invalidation
We automatically invalidate affected pixels when using pointcloud, but not alignment.

wangsff, 2020-10-18:
@dorodnic thank you for response, I have read this document before. So is there no way to get the well aligned images?
I also try another method `rs2_deproject_pixel_to_point`, `rs2_transform_point_to_point`,`rs2_project_point_to_pixel` as mentioned in #7581 . Using this method, the "duplication region" has gone, but  the aligned color map seems not to align very well, and seems to downward shift a couple of pixels to correct position.
Is that my camera Intrinsics or Extrinsics problem? How can I solve it?

dorodnic, 2020-10-19:
Hi @wangsff 
Is it viable for you to align to RGB? It should not have this artefact. 
Alternatively, you can generate `points` object using pointcloud class and use it as mask (only keep pixel i,j if points[i,j] != 0,0,0)

wangsff, 2020-10-23:
If I align to RGB, the depth map will appear many black holes. see my issue #7541

wangsff, 2020-10-23:
And I dont understand what you mean by `use it as mask (only keep pixel i,j if points[i,j] != 0,0,0`.
The pixel you mention means color pixel? @dorodnic

anhTuan0712, 2020-12-03:
> I believe its effect of occlusion - https://dev.intelrealsense.com/docs/projection-texture-mapping-and-occlusion-with-intel-realsense-depth-cameras#section-4-occlusion-invalidation
> We automatically invalidate affected pixels when using pointcloud, but not alignment.

Hello could you help me to solve this problem, I expect feedback from you, thank you! https://github.com/IntelRealSense/librealsense/issues/7906

RealSenseSupport, 2021-03-24:
Hi,

Do you still need help with this? 

Curious if enabling the 'Invalidation Bypass' sensor option (via rs-sensor-control tool) in the L515 helps? The setting will persist in the camera.

Thanks

closed 2021-04-02 by RealSenseSupport