realsenseai/librealsense #7561: multicam pointcloud

wuhaoqi-hub, 2020-10-13:
I want to control two cameras to get the pointcloud ,then save it as ply format,i also want to control the exposure and the emitter,Can you recommend some useful examples for me to learn? Thank you very much.

MartyG-RealSense, 2020-10-13:
Hi @wuhaoqi-hub   If you plan to control these functions by writing your own program, could you tell me please which programming language you are using please (C++, C#, Python), etc.

wuhaoqi-hub, 2020-10-14:
c++------------------&nbsp;原始邮件&nbsp;------------------
发件人:&nbsp;&quot;MartyG-RealSense&quot;<notifications@github.com&gt;
发送时间:&nbsp;2020年10月13日(星期二) 晚上11:04
收件人:&nbsp;&quot;IntelRealSense/librealsense&quot;<librealsense@noreply.github.com&gt;;
抄送:&nbsp;&quot;wuhaoqi-hub&quot;<742842498@qq.com&gt;;&quot;Mention&quot;<mention@noreply.github.com&gt;;
主题:&nbsp;Re: [IntelRealSense/librealsense] multicam pointcloud (#7561)

wuhaoqi-hub, 2020-10-14:
c++

MartyG-RealSense, 2020-10-14:
Thank you very much for the confirmation that you are using C++.  

Some links to programming information for multicam point cloud, emitter and exposure are provided below.  I have listed one example link for each type of function.  Please let me know if you need further examples after studying these.

**MULTICAM POINT CLOUD MERGING**

https://github.com/IntelRealSense/librealsense/issues/2556

**EMITTER ON / OFF**

Code for controlling the emitter on / off state with the **RS"_OPTION_EMITTER_ENABLED** instruction can be found in the section headed **librealsese2** in this link:

https://github.com/IntelRealSense/librealsense/wiki/API-How-To#controlling-the-laser

**MANUAL EXPOSURE**

There are separate exposure controls for depth and RGB sensors.

**RGB exposure**
https://github.com/IntelRealSense/librealsense/issues/3224

**Depth exposure**
https://github.com/IntelRealSense/librealsense/issues/5528

wuhaoqi-hub, 2020-10-15:
thank you very much!

closed 2020-10-15 by wuhaoqi-hub