realsenseai/librealsense #7562: On-Chip calibration does not update calibration table - values do not differ from 'gold' factory values.

Chandler, 2020-10-13:
Steps to reproduce with a d415 running on ubuntu 18

1) reset camera to gold factory calibration
> `CustomRW -g`

2) save calibration table
`CustomRW -r -f gold.xml`

3) run `rs-depth-quality` > more > on chip calibration > intrinsics|extrinsics > white wall > apply new

4) dump new calibration table `CustomRW -r -f new.xml`

5) observe that new.xml and gold.xml are identical

MartyG-RealSense, 2020-10-14:
Hi @Chandler  The -r option allows you to **read** the calibration parameters currently stored in a device, and save them as a backup to an .xml file.  It is not saving the calibration parameters into the device.  To save parameters to the device, -w is used instead of -r.

My interpretation of the procedure that you described is that you are not using the Dynamic Calibrator to perform a re-calibration.  Instead, you are exporting the calibration settings as they **currently** are into an xml file.

Then you perform an On-Chip calibration, and use CustomRW to export the current calibration settings again to a new filename.  When you compare the two files their calibration details are identical, suggesting that either the On-Chip calibration process did nothing or that the changes were not burned to the camera when the **Apply New** button was clicked.  Is my interpretation correct, please?

Chandler, 2020-10-14:
@MartyG-RealSense you are correct, using `rs-depth-quality` and apply new does not burn new values into the calibration table. I tested this on a d415 and d455. I'd be curious to know if you can reproduce this.

I cannot try with `realsense-viewer` because the on-chip calibration routine freezes when I use that tool.

Chandler, 2020-10-15:
update! I was able to get calibration to "do something" e.g. produce a slight calibration change via `realsense-viewer`

Steps:
realsense-viewer > more > calibration data > factory reset & save as gold.json
realsense-viewer > more > on chip calibration > extrinsics & medium > apply
realsense-viewer > more > calibration data > factory reset & save as new.json

A slight difference is found, indicating that calibration had an effect. 

$ diff gold.json new.json
[image]

I'm not sure why my earlier sequence doesn't work, perhaps because I was using white wall which I now know is not the recommended calibration mode.

Chandler, 2020-10-15:
I think the issue may have been that I was using realsense viewer 2.34 and in the follow up test I upgraded to 2.38

MartyG-RealSense, 2020-10-15:
It may indeed have been due to SDK 2.34, as I recall that this version had problems with the On-Chip Calibration function.

https://github.com/IntelRealSense/librealsense/issues/6231

Great to hear that you resolved your calibration problem by changing versions!

closed 2020-10-15 by Chandler