realsenseai/librealsense #7565: L515: General information/direction on handheld 3D Scanning

abribanez, 2020-10-14:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC |
| Language                            |  Python                          | 

### Issue Description
Hi, 
I'm looking for further understanding regarding how the process of 3D Scanning will work in my case and what extra software I could possibly need.
I'm particularly interested in scanning interiors and ultimately get an OBJ which I can use for 3D Rendering. My understanding of 3D scanning is very low. I come from a VFX background (3D rendering, modelling, etc..)

Could you describe what the pipeline would look like once I get an L515?

Specifically:
- If I had a massive USB 3.0 cable connected from the PC to the camera (not ideal, of course) can IntelRealSense Viewer record, export and/or merge all the pointcloud data? I would imagine I'll need to process the point cloud data in another software to convert it to a 3D Mesh (OBJ format or other). If you could provide more detailed information as to how this process would look like, I would appreciate it.

Also if there's any free software solution you can recommend, that would be great, but I do have some python knowledge. Perhaps, if you could provide pointers to know how to accomplish what I want using the SDK, that would be very helpful too.  

Thank you!
Abraham I.

shuixing85, 2020-10-14:
Hi Abraham,

I have the same interest, just bought the L515 yesterday and played for a while, here are my comments:

Assuming you have a long enough cable to connect the PC:
1. download Dot3D, that will generate a ply file.
2. using Meshlab to convert the ply file to obj;
then you may play with it in the Blender or Cinema4D  : )

hope this will help you.

abribanez, 2020-10-14:
Hi! 
Thanks for the reply. Yeah, after looking at Dot3D videos, it seems to do what I'm looking after. However, it is quite expensive! hehe. I would prefer an open-source approach, even if it means having to write some code to make it work, just want some kind of direction really.

Having said that, am I right in assuming that IntelRealSense Viewer has no way to export a .ply file? (Need to do some research on what the ply file really is,  hehe, I'm very ignorant in this topic.. for now :P )

I also see there's an android app called [RS Camera](https://play.google.com/store/apps/details?id=com.intel.realsense.camera&hl=en&gl=US) - wondering what the file format coming out of those depth/rgb recordings is.

shuixing85, 2020-10-17:
Hi there,
New findings:
1. For each record, the realsense viewer save "bag" files under the document folder, which include all data generated during the record;
2. the SDK2 provides a converter tool under C:\Program Files (x86)\Intel RealSense SDK 2.0\tools, can help you to transfer the Bag file to PLYs;
3. My next step is to find some align software to place the ply models at the right places, 

Seems not far from our goal and hope it helps,

William

abribanez, 2020-10-21:
Hi William,

I came across Open3D, its open-source, and it might do what we are looking for. Although the record functionality seems not supported yet for L515, BUT reading through the ISSUES section, it could be solved by properly configuring Open3D specifically for L515:[Read this](https://github.com/intel-isl/Open3D/issues/1990)

Another solution might be to process the .ply files converted from the .bag in Open3D by using the alignment and reconstruction functions.

I haven't bought the camera yet, still deciding if I should go for it or not. Looks like there's not much support for the L515 :P I would be interested to see an OBJ sample from an L515 3D reconstruction interior scene. If you ever happen to successfully create one, would you mind sharing it with me?

Thanks for sharing what you know so far!
Abraham I.

shuixing85, 2020-10-24:
Hi Abribanez,

Yeah, I've looked at Open3D these days, literally 0 experience in coding and it progresses very slowly.
Just managed to run a few tutorials and I believe we are heading to the right direction.

I used Dot3D to create an OBJ as a sample for you, hope it helps : )
OBJ file: https://www.dropbox.com/t/hvVKFz1O6Iiqz0G2

Regards,
William

rlabs-oss, 2020-10-26:
RTAB-Map and the L515 @ 640x480 works providing you can build the standalone app and also librealsense2. 
and apply the suggested changes from the [RTAB-Map issue #614](https://github.com/introlab/rtabmap/issues/614) 
A short demo[ mapping with the L515](https://www.youtube.com/watch?v=iVVb01PIngo) 
I did test the L515 in IR mode at 1024x768 also but that causes problems with the other T200/D400 cameras.

shuixing85, 2020-10-31:
> RTAB-Map and the L515 @ 640x480 works providing you can build the standalone app and also librealsense2.
> and apply the suggested changes from the [RTAB-Map issue [...]

@rlabs-oss Thanks for helping, it is exactly what I want, 

Happy to see someone has figured it out so I can just follow the steps : )

shuixing85, 2020-12-02:
Hi @abribanez

I tested/tried RTABmap these days/finally managed to install on my machine,

and it works similar to the Dot3D, the quality may differ a little bit, too many parameters to tweak, but in general, it works!

You may get the package from the building frame here: https://ci.appveyor.com/project/matlabbe/rtabmap/branch/master/artifacts

turn off the globe time sync as this:
https://github.com/introlab/rtabmap/issues/614#issuecomment-734101074

And it should work.

Hope it helps you,

William

eltonhcchan, 2021-01-16:
@rlabs-oss, I am also trying L515 but on windows 10. I followed your advice and can show input streaming when testing only. The start button is disabled but I don't know how to activate it. Any advice?

rlabs-oss, 2021-01-16:
File / New Database, then click start - See video example (This was made in Linux, should be the same for Windows)

https://user-images.githubusercontent.com/72981257/104816846-5ac82600-5826-11eb-9ca9-c61aed2fdbf8.mp4

eltonhcchan, 2021-01-17:
@rlabs-oss many thx, it finally works. BTW, it is easy to lose track.

closed 2021-02-12 by RealSenseSupport