realsenseai/librealsense #7566: Ubuntu 20.04.1 LTS focal-raspi-5.4 patch help

stephen-webb, 2020-10-14:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  L515, R200
| Firmware Version                   | 
| Operating System & Version |   Linux  - Ubuntu 20.04 
| Kernel Version (Linux Only)    |  Ubuntu 20.04.1 Linux 5.4-raspi    
| Platform                                 |  Raspberry Pi
| SDK Version                           |  2.38.1
| Language                               |  C++
| Segment		   	         |  Robot

### Issue Description
Hi, Would someone be able to suggest how I could move forward with my attempt to patch Ubuntu 20.04 modules?

I have developed and applied the patches to videodev, uvcvideo, hid_sensor_accel_3d and hid_sensor_gyro_3d for the linux-raspi-5.4.0-1019 code. (See https://github.com/IntelRealSense/librealsense/compare/master...stephen-webb:ubuntu_20_04_raspi)

Using librealsense2.so.2.38.1 without the patched modules resulted in:
  2020-10-14 10:52:34,772 [0xffffb304e1b0] DEBUG libRealsense - Found 0 RealSense devices (mask 254)

Using librealsense2.so.2.38.1 with the patched modules resulted in:
  2020-10-14 10:52:34,764 [0xffffb304e1b0] DEBUG libRealsense - Failed to read busnum/devnum. Device Path: /sys/dev/char/81:0/device/../../../../../../../../../../
  2020-10-14 10:52:34,772 [0xffffb304e1b0] DEBUG libRealsense - Found 0 RealSense devices (mask 254)
  2020-10-14 10:52:34,773 [0xffffb304e1b0] DEBUG libRealsense - Failed to read busnum/devnum. Device Path: /sys/dev/char/81:4/device/../../../../../../../../../../
  2020-10-14 10:52:34,774 [0xffffb304e1b0] DEBUG libRealsense - Failed to read busnum/devnum. Device Path: /sys/dev/char/81:1/device/../../../../../../../../../../
  2020-10-14 10:52:34,774 [0xffffb304e1b0] DEBUG libRealsense - Failed to read busnum/devnum. Device Path: /sys/dev/char/81:2/device/../../../../../../../../../../
  2020-10-14 10:52:34,775 [0xffffb304e1b0] DEBUG libRealsense - Failed to read busnum/devnum. Device Path: /sys/dev/char/81:5/device/../../../../../../../../../../
  2020-10-14 10:52:34,776 [0xffffb304e1b0] DEBUG libRealsense - Failed to read busnum/devnum. Device Path: /sys/dev/char/81:3/device/../../../../../../../../../../
  2020-10-14 10:52:34,776 [0xffffb304e1b0] DEBUG libRealsense - Failed to read busnum/devnum. Device Path: /sys/dev/char/81:6/device/../../../../../../../../../../
  2020-10-14 10:52:34,777 [0xffffb304e1b0] DEBUG libRealsense - Failed to read busnum/devnum. Device Path: /sys/dev/char/81:0/device/../../../../../../../../../../
  2020-10-14 10:52:34,785 [0xffffb304e1b0] DEBUG libRealsense - Found 0 RealSense devices (mask 254)

So it seems to have progressed towards a result but some patch is still missing.

Here is the system log with UVC_TRACE_CONTROL bit in the module parameter ‘trace’ enabled. The initial entries are from connecting the R200. Later entries are from connecting the L515:

[Wed Oct 14 08:33:18 2020] usb 1-1.1: new high-speed USB device number 5 using xhci_hcd
[Wed Oct 14 08:33:18 2020] usb 1-1.1: New USB device found, idVendor=8086, idProduct=0b64, bcdDevice=10.41
[Wed Oct 14 08:33:18 2020] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Wed Oct 14 08:33:18 2020] usb 1-1.1: Product: Intel(R) RealSense(TM) 515
[Wed Oct 14 08:33:18 2020] usb 1-1.1: Manufacturer: Intel(R) RealSense(TM) Camera
[Wed Oct 14 08:33:18 2020] usb 1-1.1: SerialNumber: 00000000F0221034
[Wed Oct 14 12:48:58 2020] usb 2-2: new SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[Wed Oct 14 12:48:58 2020] usb 2-2: New USB device found, idVendor=8086, idProduct=0a80, bcdDevice=14.84
[Wed Oct 14 12:48:58 2020] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Wed Oct 14 12:48:58 2020] usb 2-2: Product: Intel RealSense 3D Camera R200
[Wed Oct 14 12:48:58 2020] usb 2-2: Manufacturer: Intel Corp
[Wed Oct 14 12:48:58 2020] usb 2-2: SerialNumber: SN_2511006763
[Wed Oct 14 12:48:58 2020] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[Wed Oct 14 12:48:58 2020] uvcvideo 2-2:1.0: Entity type for entity Extension 2 was not initialized!
[Wed Oct 14 12:48:58 2020] uvcvideo 2-2:1.0: Entity type for entity Camera 1 was not initialized!
[Wed Oct 14 12:48:58 2020] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[Wed Oct 14 12:48:58 2020] uvcvideo 2-2:1.2: Entity type for entity Camera 4 was not initialized!
[Wed Oct 14 12:48:58 2020] input: Intel RealSense 3D Camera R200: as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-2/2-2:1.2/input/input0
[Wed Oct 14 12:48:58 2020] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[Wed Oct 14 12:48:58 2020] uvcvideo: Added control 00000000-0000-0000-0000-000000000001/2 to device 2 entity 6
[Wed Oct 14 12:48:58 2020] uvcvideo: Adding mapping 'Exposure, Auto' to control 00000000-0000-0000-0000-000000000001/2.
[Wed Oct 14 12:48:58 2020] uvcvideo: Added control 00000000-0000-0000-0000-000000000001/4 to device 2 entity 6
[Wed Oct 14 12:48:58 2020] uvcvideo: Adding mapping 'Exposure (Absolute)' to control 00000000-0000-0000-0000-000000000001/4.
[Wed Oct 14 12:48:58 2020] uvcvideo: Added control 00000000-0000-0000-0000-000000000101/2 to device 2 entity 7
[Wed Oct 14 12:48:58 2020] uvcvideo: Adding mapping 'Brightness' to control 00000000-0000-0000-0000-000000000101/2.
[Wed Oct 14 12:48:58 2020] uvcvideo: Added control 00000000-0000-0000-0000-000000000101/3 to device 2 entity 7
[Wed Oct 14 12:48:58 2020] uvcvideo: Adding mapping 'Contrast' to control 00000000-0000-0000-0000-000000000101/3.
[Wed Oct 14 12:48:58 2020] uvcvideo: Added control 00000000-0000-0000-0000-000000000101/6 to device 2 entity 7
[Wed Oct 14 12:48:58 2020] uvcvideo: Adding mapping 'Hue' to control 00000000-0000-0000-0000-000000000101/6.
[Wed Oct 14 12:48:58 2020] uvcvideo: Added control 00000000-0000-0000-0000-000000000101/7 to device 2 entity 7
[Wed Oct 14 12:48:58 2020] uvcvideo: Adding mapping 'Saturation' to control 00000000-0000-0000-0000-000000000101/7.
[Wed Oct 14 12:48:58 2020] uvcvideo: Added control 00000000-0000-0000-0000-000000000101/8 to device 2 entity 7
[Wed Oct 14 12:48:58 2020] uvcvideo: Adding mapping 'Sharpness' to control 00000000-0000-0000-0000-000000000101/8.
[Wed Oct 14 12:48:58 2020] uvcvideo: Added control 00000000-0000-0000-0000-000000000101/9 to device 2 entity 7
[Wed Oct 14 12:48:58 2020] uvcvideo: Adding mapping 'Gamma' to control 00000000-0000-0000-0000-000000000101/9.
[... 6 lines cut ...]
[Wed Oct 14 12:48:58 2020] uvcvideo: Added control 00000000-0000-0000-0000-000000000101/5 to device 2 entity 7
[Wed Oct 14 12:48:58 2020] uvcvideo: Adding mapping 'Power Line Frequency' to control 00000000-0000-0000-0000-000000000101/5.
[Wed Oct 14 12:48:58 2020] uvcvideo: Added control 00000000-0000-0000-0000-000000000101/11 to device 2 entity 7
[Wed Oct 14 12:48:58 2020] uvcvideo: Adding mapping 'White Balance Temperature, Auto' to control 00000000-0000-0000-0000-000000000101/11.
[Wed Oct 14 12:48:58 2020] uvcvideo 2-2:1.4: Entity type for entity Processing 7 was not initialized!
[Wed Oct 14 12:48:58 2020] uvcvideo 2-2:1.4: Entity type for entity Camera 6 was not initialized!
[Wed Oct 14 12:48:58 2020] input: Intel RealSense 3D Camera R200: as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-2/2-2:1.4/input/input1
[Wed Oct 14 13:00:18 2020] usb 1-1.1: USB disconnect, device number 5
[Wed Oct 14 13:00:18 2020] usb 1-1.1: new high-speed USB device number 6 using xhci_hcd
[Wed Oct 14 13:00:18 2020] usb 1-1.1: New USB device found, idVendor=8086, idProduct=0b64, bcdDevice=10.41
[Wed Oct 14 13:00:18 2020] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Wed Oct 14 13:00:18 2020] usb 1-1.1: Product: Intel(R) RealSense(TM) 515
[Wed Oct 14 13:00:18 2020] usb 1-1.1: Manufacturer: Intel(R) RealSense(TM) Camera
[Wed Oct 14 13:00:18 2020] usb 1-1.1: SerialNumber: 00000000F0221034
[Wed Oct 14 13:00:19 2020] usb 1-1.1: USB disconnect, device number 6
[Wed Oct 14 13:00:26 2020] usb 1-1.1: new high-speed USB device number 7 using xhci_hcd
[Wed Oct 14 13:00:26 2020] usb 1-1.1: New USB device found, idVendor=8086, idProduct=0b64, bcdDevice=10.41
[Wed Oct 14 13:00:26 2020] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Wed Oct 14 13:00:26 2020] usb 1-1.1: Product: Intel(R) RealSense(TM) 515
[Wed Oct 14 13:00:26 2020] usb 1-1.1: Manufacturer: Intel(R) RealSense(TM) Camera
[Wed Oct 14 13:00:26 2020] usb 1-1.1: SerialNumber: 00000000F0221034

I have checked that the other patches in the scripts directory are not needed (the code already has those changes).

Thanks

stephen-webb, 2020-10-14:
I found the problem was a somewhat faulty USB3 cable. 

Using a different USB3 cable with the patched kernel modules now past the "Failed to read busnum/devnum." messages to:
2020-10-15 08:25:12,519 [0xffff9d6171b0] TRACE libRealsense - Recognized pixel-format 8-bit Greyscale
2020-10-15 08:25:12,519 [0xffff9d6171b0] TRACE libRealsense - Recognized pixel-format 8-bit Greyscale
2020-10-15 08:25:12,519 [0xffff9d6171b0] TRACE libRealsense - Recognized pixel-format 16-bit Depth
2020-10-15 08:25:12,519 [0xffff9d6171b0] TRACE libRealsense - Recognized pixel-format 4-bit Depth Confidence (Packed)
2020-10-15 08:25:12,524 [0xffff9d6171b0] TRACE libRealsense - Recognized pixel-format YUYV 4:2:2
2020-10-15 08:25:12,529 [0xffff9d6171b0] DEBUG libRealsense - Request: Y8 Infrared,
Resolved to: Y8 Infrared,
2020-10-15 08:25:12,531 [0xffff9d6171b0] TRACE libRealsense - Recognized pixel-format 8-bit Greyscale
2020-10-15 08:25:12,531 [0xffff9d6171b0] TRACE libRealsense - Recognized pixel-format 8-bit Greyscale
2020-10-15 08:25:12,532 [0xffff9d6171b0] DEBUG libRealsense - Video node was successfully configured to GREY format, fd 26
2020-10-15 08:25:12,532 [0xffff9d6171b0] DEBUG libRealsense - Trying to configure fourcc GREY
2020-10-15 08:25:12,532 [0xffff9d6171b0] DEBUG libRealsense - Metadata node was successfully configured to D4XX format, fd 29
2020-10-15 08:25:12,536 [0xffff9d6171b0] TRACE libRealsense - time_diff_keeper::start: _users_count = 1
2

This issue can now be closed

RealSenseCustomerSupport, 2020-10-16:
----------------------------------------------

@stephen-webb Thanks for your update!

closed 2020-10-16 by RealSenseCustomerSupport