realsenseai/librealsense #7567: L515 python get object dimensions with just one camera

leemerlas, 2020-10-14:
Hi, is it possible to get the dimensions of an object with just one camera? Asking because we ordered only one L515 Lidar camera. 

I found the `box_dimensioner_multicam_demo.py` demo [here](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/box_dimensioner_multicam/box_dimensioner_multicam_demo.py) and maybe it is possible to modify this code since only one camera will be used.

Any help/advice is appreciated, thank you.

dorodnic, 2020-10-19:
Hi @leemerlas 
All sample code is provided as reference. The box dimensioning demo is very specific to two *stereo* cameras. One L515 would require different approach, and two L515 would interfere with each other in this setup, unless you implement hardware sync.

leemerlas, 2020-10-20:
Hi @dorodnic 

Thanks for the response. Could you point me in the right direction on where to start with regards to implementing this different approach? Maybe some sample code is available?

RealSenseSupport, 2020-12-23:
@leemerlas 

If you referring to HW sync of L515 cameras, here is a description of that:  https://www.intelrealsense.com/lidar-camera-l515-multi-camera-configuration/

We also provide a box dimensioning app:  https://store.intelrealsense.com/dim-weight-software-free-trial.html

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2021-01-11 by RealSenseSupport

zeynepkoyun, 2021-04-29:
@code-trekker Hi, I want to find the object dimensions with the same camera, could you?