realsenseai/librealsense #7571: [D455] Calibrating the IMU for D455

zc08, 2020-10-14:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 05.12.06.00 | 
| Operating System & Version |  Ubuntu 18.04.3  | 
| Kernel Version (Linux Only)    |  5.4.0-48-generic                                         | 
| Platform                                 | PC  |
| SDK Version                            | 2.36.0                       | 
| Language                            |  C/python                          | 
| Segment			| Robot                         | 

### Issue Description
I found the imu calibraition tool for D435i and L515 in:
https://github.com/IntelRealSense/librealsense/tree/development/tools/rs-imu-calibration#rs-imu-calibration-tool

is it equally applicable to D455? Thanks in advance!

dorodnic, 2020-10-14:
Hi @zc08 
Yes

MartyG-RealSense, 2020-10-14:
Hi @dorodnic  Thanks very much for your answer to @zc08   👍

wangsff, 2020-10-14:
Could you please help figure out the "strange duplication region"?  #7556 It's very important to me. 
thanks in advance! @MartyG-RealSense @dorodnic

MartyG-RealSense, 2020-10-14:
Hi @wangsff  I am not involved in handling L515 cases as another RealSense team member is responsible for those.

closed 2020-10-15 by zc08