realsenseai/librealsense #7574: [D455] Depth reduced to only 4 meters on the SDK

rafaelRinconFo, 2020-10-14:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D455 } | 
| Firmware Version                   | 05.12.07.100 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  5.4.0-48-generic            | 
| Platform                                 | PC  |
| SDK Version                            |  2.38.1                        | 
| Language                            |  Python / C++                   | 
| Segment			|  Robot                        | 

Currently, my team is testing the new Intel Realsense 455, as you can supose by the maximum depth showed by the SDK is only 4m. I have read previous Issues from the 435i and other cameras (like this one [https://github.com/IntelRealSense/librealsense/issues/3139](url) ) where you state that this is due to the fact that, from 3m, the RMS error increases.

However, while a 435i camera has a depth range of 10 m and, 4m is acceptable, the 455 model is supposed to be capable of getting 20m in-depth. Losing 16m is a crippling loss in data and, we were expecting to use at least 8m.

Is there a way to increase the depth visualized on the SDK?

Furthermore, we expect to use the camera on ROS with your ROS2 wrapper (I have to say that the support of said wrapper is currently, a little bit slow), and we would like to know as well if this depth-problem is going to be present in ROS too. 

Thank you for your support.

MartyG-RealSense, 2020-10-14:
Hi @rafaelRinconFo   If your depth is being capped at 4 meters whilst testing with the RealSense Viewer program, this suggests that the **threshold filter** option in the **Post-Processing** section of the Viewer's options side-panel may be enabled, as its default maximum range setting is '4' (meters).  When the threshold filter is enabled, depth values outside of the defined minimum and maximum will be excluded from the image.

Regarding ROS2 support for RealSense: the official ROS2 wrapper is currently new at the time of writing this and will develop over time like the ROS1 wrapper did.  librealsense and the ROS wrapper operate independently though, with settings made in librealsense not affecting the camera in the ROS wrapper and vice versa.  The ROS wrapper also lacks a threshold filter and instead uses an option called **clip_distance** to set a maximum distance.

rafaelRinconFo, 2020-10-14:
Hi @MartyG-RealSense , thank you so much for your fast response. Indeed, this option had to be set and now, everything works perfectly. 

Regarding the ROS subject, I'm going to address the ROS2 developers then.

Again, thank you so much for your fast and clear response.

MartyG-RealSense, 2020-10-14:
You are very welcome.   I'm pleased that I could help!

closed 2020-10-15 by rafaelRinconFo