realsenseai/librealsense #7575: Mistype in Intel® RealSense™ D400 Series Calibration Tools User Guide

TDHster, 2020-10-14:
Trying to make dynamic calibration on Ubuntu 20.04 with recommendation from: https://github.com/IntelRealSense/librealsense/issues/2329

> Instructions for installing the Linux version of the tool are on page 14 onward of the user guide for the tool.
> 
> https://www.intel.com/content/www/us/en/support/articles/000026723/emerging-technologies/intel-realsense-technology.html

$ echo 'deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionicmain' | sudo tee /etc/apt/sources.list.d/realsense-public.list deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionicmain
$ sudo apt-key adv --keyserver keys.gnupg.net --recv-key 6F3EFCDE
Executing: /tmp/apt-key-gpghome.1FoxtUlaPC/gpg.1.sh --keyserver keys.gnupg.net --recv-key 6F3EFCDE
gpg: key C8B3A55A6F3EFCDE: ""CN = Intel(R) Intel(R) Realsense", O=Intel Corporation" not changed
gpg: Total number processed: 1
gpg:              unchanged: 1
$ sudo apt update
E: Malformed entry 1 in list file /etc/apt/sources.list.d/realsense-public.list (Component)
E: The list of sources could not be read.

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Operating System & Version |   Ubuntu 20.04 LTS | 
| Kernel Version (Linux Only)    | 5.4.0-51-generic                                        | 
| Platform                                 | PC |

### Issue Description
Because copied from PDF instruction doesn't have a space between "bionic" and "main" and it's copied as "bionicmain" in file that cause error.

MartyG-RealSense, 2020-10-14:
Hi @TDHster   I tried to recreate how this may have happened, as the sentence is written correctly as **bionic main** in the PDF of the Dynamic Calibrator.   Corruptions can happen if you try to copy and paste lines from the pages of a PDF, and this is common with PDF files in general.  So the content of the document is correct, I believe, and it is just the **structure** of the file that makes it difficult to directly copy text from.

TDHster, 2020-10-18:
Thank  you, MartyG.

closed 2020-10-18 by TDHster