realsenseai/librealsense #7578: LNK2019,无法解析外部代码

zhangcongyao0921, 2020-10-15:
严重性	代码	说明	项目	项目级别	文件	行	禁止显示状态
错误	LNK2019	无法解析的外部符号 __imp_glBegin，该符号在函数 "private: static void __cdecl imu_renderer::draw_axes(float,float)" (?draw_axes@imu_renderer@@CAXMM@Z) 中被引用	multicam	2	C:\Program Files (x86)\Intel RealSense SDK 2.0 (Win7)\test\hello\multicam\multicam.obj	1	

能够找到glbegin函数说在的GL.h文件，添加到库文件后，还是没办法解决问题，没有找到关于GL.lib文件。

MartyG-RealSense, 2020-10-15:
**Translation:** Error LNK2019 The unresolved external symbol __imp_glBegin, which is referenced in the function "private: static void __cdecl imu_renderer::draw_axes(float,float)" (?draw_axes@imu_renderer@@CAXMM@Z) multicam 2 C:\Program Files (x86)\Intel RealSense SDK 2.0 (Win7)\test\hello\multicam\multicam.obj 1

I was able to find the GL.h file where the glbegin function said, and after adding it to the library file, there was still no way to solve the problem, and no GL.lib file was found.

******************

Hello @zhangcongyao0921   If you are creating your own version of the Multicam example program and you have a LNK2019 problem with the GLFW graphics, I would recommend reading the guide in the link below.  It explains how to link to the GLFW files in VisualStudio.

https://github.com/EduardoWang/visual-studio-real-sense-record-and-playback-project

**********

您好@ zhangcongyao0921如果您正在創建自己的Multicam示例程序版本，並且GLFW圖形存在LNK2019問題，我建議您閱讀下面鏈接中的指南。 它說明瞭如何鏈接到VisualStudio中的GLFW文件。

https://github.com/EduardoWang/visual-studio-real-sense-record-and-playback-project

MartyG-RealSense, 2020-10-22:
Do you still require assistance with this case, please?   Thanks!

在這種情況下，您仍然需要幫助嗎？ 謝謝！

MartyG-RealSense, 2020-10-30:
Case closed due to no further comments received.

案件未結，因為未收到進一步評論。

closed 2020-10-30 by MartyG-RealSense