realsenseai/librealsense #7579: Kernel downgrade no longer detects cameras

fiorano10, 2020-10-15:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | (5.12.05 | 
| Operating System & Version |  Linux (Ubuntu 118.04  | 
| Kernel Version (Linux Only)    | 5.3.0.36                               | 
| Platform                                 | PC  |
| SDK Version                            |  2.33              | 
| Language                            |  C++                       | 
| Segment			|  Robot                       | 

### Issue Description

Accidentally upgraded kernel to 5.4, realsense SDK stops working, or is buggy. Tried downgrading it to 5.3.* using apt. After the downgrade `rs-fw-update -l` no longer detects realsense cameras. They show up on lsusb, they are recognized by checking `journalctl`. Tried reinstalling librealsense multiple times, but doesn't work. The only way to get them back is by reinstall the OS.

MartyG-RealSense, 2020-10-15:
Hi @fiorano10   Could you describe the process that you used to downgrade the kernel, please?  My understanding is that you should be able to do so in Ubuntu without the involvement of librealsense.  For example, one method of doing it would be to remove the newer kernel version with a **sudo apt remove** command, like the guide in the link below describes:

https://www.maketecheasier.com/downgrade-kernel-linux/

fiorano10, 2020-10-17:
Followed the exact same steps mentioned in the link. After the downgrade, librealsense no longer detects connected cameras

MartyG-RealSense, 2020-10-18:
Does the camera detectin return if you re-upgrade the kernel to 5.4 please?

closed 2020-11-02 by MartyG-RealSense