realsenseai/librealsense #7581: [L515] align offset

wangsff, 2020-10-15:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {L515  | 
| Firmware Version                   | 01.05.01.03 | 
| Operating System & Version |  win10  |                                  | 
| Platform                                 | PC |
| SDK Version                            |  realsense viewer v2.38.1  | 
| Language                            | python }                          | 

### Issue Description
As I mention on #7556 , use your align code has come up the "strange duplication region", so I want to align the color map to depth map with another menthod.
Here is code I use.
```
import pyrealsense2 as rs
import numpy as np
import cv2

config = rs.config()
config.enable_stream(rs.stream.depth, 1024, 768, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 1280, 720, rs.format.bgr8, 30)

pipeline = rs.pipeline()

pipe_profile = pipeline.start(config)

frames = pipeline.wait_for_frames()
depth_frame = frames.get_depth_frame()
color_frame = frames.get_color_frame()
# Intrinsics & Extrinsics
depth_intrin = depth_frame.profile.as_video_stream_profile().intrinsics
color_intrin = color_frame.profile.as_video_stream_profile().intrinsics
depth_to_color_extrin = depth_frame.profile.get_extrinsics_to(color_frame.profile)
color_to_depth_extrin = color_frame.profile.get_extrinsics_to(depth_frame.profile)
colorizer = rs.colorizer(2)
colorizer_depth = np.asanyarray(colorizer.colorize(depth_frame).get_data())
cv2.imwrite(r'C:\Users\DELL\Desktop\depth.png', colorizer_depth)
# color_image = np.asanyarray(color_frame.get_data())
# cv2.imwrite(r'C:\Users\DELL\Desktop\color.png', color_image)

print ("\n Depth intrinsics: " + str(depth_intrin))
print ("\n Color intrinsics: " + str(color_intrin))
print ("\n Depth to color extrinsics: " + str(depth_to_color_extrin))

# Depth scale - units of the values inside a depth frame, i.e how to convert the value to units of 1 meter
depth_sensor = pipe_profile.get_device().first_depth_sensor()
# depth_scale = depth_sensor.get_depth_scale()
depth_scale = depth_sensor.get_option(rs.option.depth_units)
depth_image = np.asanyarray(depth_frame.get_data())
print ("\n\t depth_scale: " + str(depth_scale))
color_image = np.asanyarray(color_frame.get_data())

height = depth_image.shape[0]
width = depth_image.shape[1]
aligned_color = np.zeros((height, width, 3))
height_color = color_image.shape[0]
width_color = color_image.shape[1]

for v in range(width_color):
    for u in range(height_color):

        color_pixel = [v, u]
        color_point = rs.rs2_deproject_pixel_to_point(color_intrin, color_pixel, 1)
        depth_point = rs.rs2_transform_point_to_point(color_to_depth_extrin, color_point)
        depth_pixel = rs.rs2_project_point_to_pixel(depth_intrin, depth_point)
        # depth_pixel = rs.rs2_project_color_pixel_to_depth_pixel(depth_frame.get_data(), depth_scale,
        #     0.11, 1.0, depth_intrin, color_intrin, depth_to_color_extrin, color_to_depth_extrin, color_pixel)
        u_depth = int(round(depth_pixel[1]))
        v_depth = int(round(depth_pixel[0]))
        if u_depth < 0 or u_depth > height-1 or v_depth < 0 or v_depth > width-1:
            pass
        else:
            aligned_color[u_depth][v_depth][0] = color_image[u][v][0]
            aligned_color[u_depth][v_depth][1] = color_image[u][v][1]
            aligned_color[u_depth][v_depth][2] = color_image[u][v][2]
cv2.imwrite(r'C:\Users\DELL\Desktop\aligned_color.png', aligned_color)

```
And the aligned color map seems not to align very well.

aligned color map
[image]
origin depth map
[image]

You can the the aligned color map downward shift to the correct position. 
How can I solve it? Thanks in advance!

cdb0y511, 2020-11-19:
Hi, @wangsff 
This may be related to #7335 
The Inverse-Brown-Conrady model may not be accurate.

cdb0y511, 2020-11-19:
And I wonder you may do the wrong alignment.
Intel uses a UV map to align RGB and depth not simply assuming the depth as 1.

 color_point = rs.rs2_deproject_pixel_to_point(color_intrin, color_pixel, 1)

https://dev.intelrealsense.com/docs/projection-texture-mapping-and-occlusion-with-intel-realsense-depth-cameras#33-the-uv-map

RealSenseSupport, 2021-03-21:
FYI...

https://github.com/IntelRealSense/librealsense/issues/7335#issuecomment-803628803

Thanks

closed 2021-05-15 by RealSenseSupport