realsenseai/librealsense #7582: Upgrading to SDK2 from the older Intel::RealSense SDK

GilVol, 2020-10-15:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300                                                     | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

Hi,

I am working on a robot for medical purposes. 
Up until now, I used the older SDK (for F200 and SR300). Now, i want to upgrade to SDK2 (for SR300).
1. I wanted to know if there is a difference in the accuracy in the aspect of the SDK. meaning - will i get the scanned object in a different location if I will replace the SDK?
Where can I find such documentation?

2. In the aspect of error handling, is there a documentation that can explain the possible root cause for any error that might occur, as well as how to handle in any failure case?

thanks,
Gil

MartyG-RealSense, 2020-10-15:
Hi @GilVol 

1.  Are you asking if your scan will look significantly different in appearance in librealsense2 compared to the output in librealsense1, please?  My expectation would be that the raw image wuld look similar, **but** you would have more adjustment options available in librealsense2 than you would have had in librealsense1.   The image below of an SR300 connected to librealsense2's **RealSense Viewer** program shows the settings available in its options side-panel.

2.  Librealsense has a **Troubleshooting Q&A** document.

https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A

If you encounter an error message in librealsense2, it may be best to ask a question about it at this forum or research existing cases.  The best way to get relevant search results about librealsense problems is to start every search term with the word **librealsense**, as the search engine will then be most likely to include discussions from this GitHub support forum in the search results.  For example, if you had the 'Frames didn't arrived within 5000' error then you could use this search term:

**librealsense frames didn't arrived within 5000**

GilVol, 2020-10-15:
Thank you for the quick response.
1. I mean in general will the output image will be similar to the one in the old SDK. Since it is used for medical purposes, we don't want the output to be different in terms of location of all the pixels - up to 2.5mm change. More than that can be dangerous.
2. In error handling, I meant, the known errors to be handled in runtime (For instance if the camera is off show a popup with a relevant message)

MartyG-RealSense, 2020-10-15:
1.  As I have not used librealsense1, I cannot make a precise like-for-like comparison between the outputs of librealsense1 and librealsense2 when usng SR300.  @dorodnic the RealSense SDK Manager has worked with librealsense1 and so may be able to offer you advice on this point.

2.  Error and event messages most commonly appear as text messages in a log during runtime.  In the RealSense Viewer, you can toggle a log message window at the bottom of the program's window as visible or hidden.

GilVol, 2020-10-15:
Hi,

I am referring to code that is running via C++. The exceptions available are:
typedef enum rs2_exception_type
{
    RS2_EXCEPTION_TYPE_UNKNOWN,
    RS2_EXCEPTION_TYPE_CAMERA_DISCONNECTED,      /**< Device was disconnected, this can be caused by outside intervention, by internal firmware error or due to insufficient power */
    RS2_EXCEPTION_TYPE_BACKEND,                  /**< Error was returned from the underlying OS-specific layer */
    RS2_EXCEPTION_TYPE_INVALID_VALUE,            /**< Invalid value was passed to the API */
    RS2_EXCEPTION_TYPE_WRONG_API_CALL_SEQUENCE,  /**< Function precondition was violated */
    RS2_EXCEPTION_TYPE_NOT_IMPLEMENTED,          /**< The method is not implemented at this point */
    RS2_EXCEPTION_TYPE_DEVICE_IN_RECOVERY_MODE,  /**< Device is in recovery mode and might require firmware update */
    RS2_EXCEPTION_TYPE_IO,                       /**< IO Device failure */
    RS2_EXCEPTION_TYPE_COUNT                     /**< Number of enumeration values. Not a valid input: intended to be used in for-loops. */
} rs2_exception_type;

Thanks,
Gil

MartyG-RealSense, 2020-10-15:
Aside from that full list of exceptions from the documentation, you may also find the information in the link below useful in regard to error handling and exception catching.

https://github.com/IntelRealSense/librealsense/issues/7513

GilVol, 2020-10-19:
Thank you, that answers my 2nd question.
any updates on the 1st question?

MartyG-RealSense, 2020-10-19:
_I wanted to know if there is a difference in the accuracy in the aspect of the SDK. meaning - will i get the scanned object in a different location if I will replace the SDK?
Where can I find such documentation?_

_I mean in general will the output image will be similar to the one in the old SDK. Since it is used for medical purposes, we don't want the output to be different in terms of location of all the pixels - up to 2.5mm change. More than that can be dangerous._

@dorodnic or @ev-mp   Can you help @GilVol with their question please about whether they will see significantly different results in their SR300 image in librealsense2 compared to the results they got from librealsense1 please?   Thanks!

MartyG-RealSense, 2020-10-27:
Hi @GilVol   Do you require further assistance with this case, please?   Thanks!

GilVol, 2020-10-27:
yes, I am still waiting for @dorodnic  or @ev-mp  response.

MartyG-RealSense, 2020-10-27:
It occurred to me that one way to help answer your question may be to look at how librealsense1 and librealsense2 handle the **projection** of the image.  I found that the projection configuration details for both SDKs are identical.

**librealsense1**
https://github.com/IntelRealSense/librealsense/blob/v1.12.1/doc/projection.md#f200-and-sr300

**librealsense2**
https://github.com/IntelRealSense/librealsense/wiki/Projection-in-RealSense-SDK-2.0#sr300

GilVol, 2020-10-27:
Thanks!
you can close the issue.

MartyG-RealSense, 2020-10-27:
Thanks very much - I am glad I could be of help.  I will close the case now.  Good luck!

closed 2020-10-27 by MartyG-RealSense