realsenseai/librealsense #7583: mstsc can't open viewer

wuhaoqi-hub, 2020-10-15:
When I use mstsc remote desktop of win7 system, I can't open the Intel realsense viewer application. If I don't use remote connection, it can work normally. Has anyone encountered this problem before? How to solve it?

MartyG-RealSense, 2020-10-15:
Hi @wuhaoqi-hub  I do not know of past cases where the Viewer has been connected to using mstsc.  The Viewer has successfully been connected to with **Chrome Remote Desktop** in the past though, at least in 2D mode.  

https://github.com/IntelRealSense/librealsense/issues/5217

Would using Chrome Remote Desktop instead of mstsc be a possibility for you, please?

https://support.google.com/chrome/answer/1649523?co=GENIE.Platform%3DDesktop&hl=en

dorodnic, 2020-10-18:
Hi @wuhaoqi-hub 
There is a known issue (in Windows) with running OpenGL applications over MSTSC session. I can propose using VNC or looking online for work-around for Win7. This is a general OS issue, not strictly related to RealSense

closed 2020-10-27 by wuhaoqi-hub