realsenseai/librealsense #7584: No Frame recived issue

ranjitkathiriya, 2020-10-15:
| | |
|---|---|
|**librealsense**|2.39.0 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D435I|
|**Serial Number**|026322072776|
|**Firmware Version**|05.12.08.200|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|950323050305|
|**Firmware Update Id**|950323050305|

Please provide a description of the problem

MartyG-RealSense, 2020-10-15:
Hi @ranjitkathiriya   It looks as though you have depth, color and IMU all enabled simultaneously.  The problem that you are experiencing is a known issue that can occur when all 3 stream types are used.  There are fixes that can be programmed if creating your own applications in C++ or Python, but I am not aware of a solution to the issue for the RealSense Viewer.

In my own experience, the problem is less likely to occur if using the **rs-multicam** example program.  Despite its name, it can be used with a single camera and can display depth, color and IMU panels like the Viewer does (though non-interactive) when a D435i camera is plugged in.  

You can find a pre-built executable version of rs-multicam in the SDK's **Tools** folder.  You can find this folder easily by right-clicking on the Viewer launch icon on your Windows desktop and selecting the **Open file location** menu option.

MartyG-RealSense, 2020-10-23:
Hi @ranjitkathiriya   Do you still require assistance with this case, please?   Thanks!

ranjitkathiriya, 2020-10-23:
I have tried version 2.23 it works fine for me...

closed 2020-10-23 by ranjitkathiriya

MartyG-RealSense, 2020-10-23:
Great to hear - thanks so much for the update!