realsenseai/librealsense #7585: Depth stream start failure

pcmihnea, 2020-10-15:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.07.100 | 
| Operating System & Version |   (a)Win10Enterprise b17763 / (b)Ubuntu 18.04.04LTS  | 
| Kernel Version (Linux Only)    | (b)4.9.140  | 
| Platform                                 | (a)PC / (b)NvidiaJetsonNano |
| SDK Version                            |  (a)2.38.1 / (b)2.35.2 | 
| Language                            |  N/A  | 
| Segment			|  N/A | 

### Issue Description
Hello,

I've recently bought a Realsense D435 camera for integration on a robot in scope of obstacle avoidance, based on a Jetson Nano board.
The first step I took was to test it using the Realsense Viewer from the latest SDK release for Windows.
Once the firmware was updated to the latest production version (05.12.07.100), I activated the two streams, stereo and RGB.
The RGB stream works flawlessly even at 1080p/30fps, but I don't receive any stereo frame.

The errors messages I get are three, continuously every few seconds:
`types.h:1099 - USB SCP overflow`
`types.h:1099 - Depth stream start failure`
`types.h:1099 - REC error`

The camera SN is 022422070xxx, ID 022523020xxx. The same HW/SW setup (USBcable/port, PC/devboard, SDK, configuration) works perfectly with another D435 camera with SN 014122072xxx, ID 015223022xxx.

I've attempted the following changes, none of which yielded any success:
-hardware: USB cable, USB type-c plug flip, USB port, USB2.0 port, PC to JetsonNano, type-C socket cleanup, temperature check (ASIC-32dgrC, emitter-30dgrC),
-software: gold factory reset, update to latest development firmware (5.12.8.200) and previous production (5.11.6.250), reflash and also recovery flash to latest production firmware (05.12.07.100), reduce/disable laser emitter, enable only stereo stream, reduce stereo stream features (resolution, processing, etc.).

Please advise on how to fix this issue. Thank you!

MartyG-RealSense, 2020-10-15:
Hi @pcmihnea   Because you mentioned UWP, as a first step in the investigation, can you confirm please whether or not you have installed the **Universal Windows Platform (UWP)** driver?

[image]

If you have installed this driver, could you uninstall it please and let us know if it solves your problem.  This driver is not required to use RealSense with Windows, as it is intended to allow a RealSense camera to interface with products that support UWP, such as the Microsoft Hololens headset.  Most RealSense users therefore do not require it.  

There have also been issues recently regarding having the latest UWP driver version **6.11.120.21** installed that are resolved if that driver is uninstalled.

pcmihnea, 2020-10-15:
Hi @MartyG-RealSense, thanks for the quick response,

The UWP issue I've observed was just a sidenote, as using the D435 camera just as a webcam is not the targeted application.
As I've mentioned in the description, another D435 camera works perfectly on the same hardware/software setup - I've even done a clean reinstall today on the Jetson Nano to rule out any OS-related issues.
If I simply replace the non-working D435 with the older one (even using the same cable), I can receive both streams - this leads me to believe the issue is with the camera.
The difference between the two camera seems to boil down to SN and FW ID's, as both were bought brand-new from a reputable supplier, with a few months between them.

MartyG-RealSense, 2020-10-15:
Thanks for the further details.  

If you uninstall the Depth driver from the Device Manager then the RGB driver should also be uninstalled.   If you then unplug the camera, wait a couple of seconds and re-insert the camera, Windows should reinstall the two drivers automatically over an internet connection.  That is the appropriate procedure for an uninstall and clean reinstall of the RGB and Depth drivers as though it is a first-time setup of the camera after purchase.

My experience is that if RealSense functions such as depth are not working but the RGB is working then the computer is treating the camera like a basic RGB webcam instead of a RealSense camera.  You should definitely not have to uninstall the RealSense RGB driver in order for RGB to function.

For the next step to test, could you unplug the camera please and then check the **Camera** section of the Device Manager to confirm that there are no hidden RealSense drivers present.  When there is no camera attached the two drivers (RGB and Depth) should not be listed in the Device Manager.  There was a case though with a RealSense user who had eight drivers in the Device Manager instead of two though because of hidden driver entries that should not have been there.

https://github.com/IntelRealSense/librealsense/issues/7185#issuecomment-688837669

pcmihnea, 2020-10-15:
I've done as you've instructed, including a SDK and driver reinstall, but the end result is the same.
To eliminate any OS-related problems, I've installed the SDK for the first time on another PC with Windows 10 b19041, and still get the same log. So far two Windows PCs and one Ubuntu board all return the same errors.

After a quick overview on the cod , "Depth stream start failure" may be a notification from the camera, and not from librealsense, so it's signalling a hardware issue?

Are there any operations to be made on the d435 besides reflash, fw recovery, and gold reset, all of which I've tried multiple times?

MartyG-RealSense, 2020-10-16:
Thanks very much for your patience.  I analysed all of the information that you provided and came to the same conclusion that it is likely to be an issue related to the hardware of that specific camera.

I gather from the supplied information that you detached the camera with the problem from the USB cable and attached the other camera to the same cable and that camera worked fine, which would seem to eliminate the USB cable as a cause of the problem.
'
I wonder if the 50-pin 'interposer' cable that joins together the **Vision Processor D4** board and the **D430 Depth Module** board inside the camera has become partially unseated somehow.   There would be no way to check this though without opening the camera up to check it.  This would invalidate the sales warranty though.  So if it is less than 30 days since you bought the camera with the problem, it may be worth considering returning it to the place of purchase for a replacement, since you have clearly tried virtually everything possible to rectify the problem.

pcmihnea, 2020-10-16:
If there is no other non-hardware-alterating solution, then I will submit a RMA to the supplier.

Thank you for your help!

closed 2020-10-16 by pcmihnea