realsenseai/librealsense #7586: Failure to use set_region_of_interest after several stream restarts.

jeffeDurand, 2020-10-15:
----------------------------------------------------------------------------------------------------

Required Info |  
-- | --
Camera Model | { D415 }
Firmware Version | (5.12.7.100)
Operating System & Version | { Ubuntu 18.04 (NVidia Jetpack 4.4)
Kernel Version (Linux Only) | (4.9)
Platform | NVIDIA Jetson (NVidia Jetpack 4.4)
SDK Version | { 2.31 }
Language | {C++ }
Segment | { others }

### Issue Description

Sometimes, when we restart several times our pipeline, we have rare happenstance of an error of setting the exposure roi over the rgb stream. Anything we doing wrong here?

Here's the error we get :

can't set roi { min_x 296, min_y 0, max_x 1659, max_y 643 } on sensor : 
hwmon command 0x75( 0 283 128 67b ) failed (response -6= Invalid parameter) (kogmod)

Here's the overall API calls that we do in a single sequence. Anything we are doing wrong here?

   ```
    rs2::context ctx_;

    auto devices = ctx_.query_devices();
    const auto count = devices.size();
    if (count == 0)
    {
        throw std::runtime_error("No realsense device found.");
    }

    auto device_restarted{false};
    ctx_.set_devices_changed_callback([&device_restarted](rs2::event_information& info) mutable {
        if (info.get_new_devices().size() != 0)
        {
            device_restarted = true;
        }
    });
    for (auto&& device : devices)
    {
        device.hardware_reset();
    }

   auto reset_timeout = std::chrono::steady_clock::now() + chrono::seconds(15);

    do 
    {
        if (std::chrono::steady_clock::now() > reset_timeout)
        {
            log_fatal("Failed to wait for device after rs2 hardware_reset.");
        }
        this_thread::sleep_for(chrono::milliseconds(100));
    } while (!device_restarted);

    rs2::pipeline pipe_;
    rs2::config cfg;
    cfg.enable_stream(RS2_STREAM_COLOR, 1920, 1080, RS2_FORMAT_RGB8, 15);
    cfg.enable_stream(RS2_STREAM_DEPTH, 424, 240, RS2_FORMAT_Z16, 15);
    pipe_.start(cfg);

    for (auto&& device : ctx_.query_devices())
    {
        const auto device_name = device.get_info(RS2_CAMERA_INFO_NAME);
        log_trace(string("starting rs2 device : ") + device_name);
        for (auto&& sensor : device.query_sensors())
        {
             //Setting options in this order for the sensors
             // Stereo module
            // RS2_OPTION_EMITTER_ENABLED 1
            // RS2_OPTION_ENABLE_AUTO_EXPOSURE 0
            // RS2_OPTION_ENABLE_AUTO_WHITE_BALANCE 0
            // RS2_OPTION_LASER_POWER 35
            
            // RGB Camera
            // RS2_OPTION_ENABLE_AUTO_EXPOSURE 1
            // RS2_OPTION_ENABLE_AUTO_WHITE_BALANCE 1
              for (const auto& options : option)
              {
              
                   if (sensor.supports(rs_option->second))
                   {
                        sensor.set_option(rs_option->second, item.value().get<float>());
                  }
            }
      }
  }
  
  for (auto&& device : ctx_.query_devices())
  {
     for (auto&& sensor : device.query_sensors())
     {
         if (sensor.is<rs2::roi_sensor>())
         {             
             const auto is_rgb = strstr(sensor.get_info(RS2_CAMERA_INFO_NAME), "RGB") != nullptr;
             const auto rs2_exposure = Local::get_exposure(is_rgb);
             try
             {   
                 sensor.as<rs2::roi_sensor>().set_region_of_interest(rs2_exposure);
             }
             catch(const rs2::error& e)
             {
                 std::stringstream ss;
                 ss << "can't set roi { " << rs2_exposure.min_x << ", " << rs2_exposure.min_y << ", " << rs2_exposure.max_x << ", " << rs2_exposure.max_y << " } on sensor : " << e.what();
                 log_error(ss.str());
              }
         }
    }
  }

```

MartyG-RealSense, 2020-10-16:
Hi @jeffeDurand   As a starting point in the investigation, may I ask please if you built librealsense with the **RSUSB** method?

jeffeDurand, 2020-10-16:
No, here's the flag I'm passing when configuring with cmake:

mkdir -p build && cd build && cmake -GNinja \
        -DBUILD_SHARED_LIBS=ON \
        -DBUILD_EXAMPLES=ON \
        -DBUILD_UNIT_TESTS=OFF \
        -DFORCE_RSUSB_BACKEND=OFF \
        -DBUILD_PYTHON_BINDINGS=OFF \
        -DBUILD_MATLAB_BINDINGS=OFF \
        -DBUILD_OPENNI2_BINDINGS=OFF \
        -DBUILD_NODEJS_BINDINGS=OFF \
        -DBUILD_CSHARP_BINDINGS=OFF \
        -DBUILD_UNITY_BINDINGS=OFF \
        -DBUILD_WITH_TM2=OFF \
        -DBUILD_WITH_CUDA=OFF \
        -DBUILD_PYTHON_DOCS=OFF \
        -DBUILD_NETWORK_DEVICE=OFF \
	-DCMAKE_CUDA_COMPILER=/usr/local/cuda/bin/nvcc \
        -DANDROID_USB_HOST_UVC=OFF

MartyG-RealSense, 2020-10-16:
Checking through your script, my personal hunch would be to focus on verifying that the sleep timing after reset is correct.  The subject is discussed in detail in the link below

https://github.com/IntelRealSense/librealsense/issues/5349

jeffeDurand, 2020-10-16:
I would have kind of expect that the set_devices_changed_callback callback would prompt when the device is really ready to fully operate. I guess we'll catch and manage these rare issues has a special case on our end. Thanks.

jeffeDurand, 2020-10-16:
After some investigating, it appears that the problem happens a lot when no set roi is done for the depth but only for rgb. When I set both, I don't get the same errors.

MartyG-RealSense, 2020-10-17:
Hi @jeffeDurand   Is setting both ROI to avoid errors a problem for your project or is it an acceptable solution, please?

logidelic, 2020-10-18:
What does setting the exposure ROI on the depth stream do?

MartyG-RealSense, 2020-10-18:
Hi @logidelic   When a ROI is defined and auto-exposure is enabled, the SDK averages the intensity of the pixels inside the ROI and tries to maintain that intensity at a defined **setpoint** value.  

Setting a separate RGB ROI does not affect the depth image, just the color image.

Point 3 of the section of Intel's camera tuning guide that I have linked to below has further details about defining a ROI in the RealSense Viewer:

https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance#section-make-sure-image-is-properly-exposed

logidelic, 2020-10-18:
Thanks for the response Marty. However, I understand what setting an exposure ROI on the RGB stream accomplishes...

My question was specifically with regard to setting this exposure ROI on the depth stream. Does it do anything at all? I'm asking because above you're suggesting that this may be a solution to the problem encountered in the OP...

Thanks again.

MartyG-RealSense, 2020-10-18:
@logidelic  The question of the effect that an auto-exposure ROI has was answered above.  

When a depth auto-exposure ROI is defined and auto-exposure is enabled, **the intensity of the pixels inside the ROI area is averaged and the auto-exposure tries to maintain that intensity at a predefined Setpoint value.**

Also, it was @jeffeDurand who stated that setting the depth ROI fixed the problem.

logidelic, 2020-10-18:
> When a depth auto-exposure ROI is defined and auto-exposure is enabled, **the intensity of the pixels inside the ROI area is averaged and the auto-exposure tries to maintain that intensity at a predefined Setpoint value.**

I guess you are suggesting that this happens for each of the stereo-pair sensors, before they get combined to extrapolate the depth, is that right? Sorry for too many question, just want to be sure I understand.

MartyG-RealSense, 2020-10-18:
@logidelic  I am not on the developer team so I do not have that information.  I researched your new question carefully but was not able to locate any further information about how the ROI functions.  I do apologize.

closed 2020-10-21 by jeffeDurand

ttencate, 2020-12-15:
@logidelic It seems that setting the exposure ROI on the "depth" sensor actually affects the exposure of the infrared sensors, as you might expect.

logidelic, 2020-12-15:
Hmm, that's somewhat interesting, but also a bit hard to understand in terms of how that should be used...