realsenseai/librealsense #7587: Build librealsense only for D415

pmirabel, 2020-10-15:
Hi,

I was wondering if there is any ooption too bild librealsense with support for only d415 support ?  
Something like :   
`cmake .. -DCMAKE_BUILD_TYPE=Release -TARGET_DEVICE=d415 `  

thank you!

MartyG-RealSense, 2020-10-16:
Hi @pmirabel  You could perform a check for the 'friendly camera name' of the camera or for its PID identity number (PID).  

**Friendly name**
RS2_CAMERA_INFO_NAME 

**PID**
RS2_CAMERA_INFO_PRODUCT_ID

The link below contains some example code that illustrates storing the friendly camera name as "Intel RealSense D415" in a string called **devName** and then checking with an If statement whether RS2_INFO_CAMERA_NAME = devName (to confirm that the attached camera is a D415).

https://github.com/IntelRealSense/librealsense/issues/2219#issuecomment-412015600

pmirabel, 2020-10-19:
I will try as soon as possible (maybe not within this month... ⏳ ) and let you know.

MartyG-RealSense, 2020-10-27:
Adding a note to keep this case open for a further time period whilst awaiting test feedback from @pmirabel at a later date.

pmirabel, 2020-10-27:
Hello,
I am sorry but I think that I missed a point.

My question was _Can I compile librealsense itself only for a specific cam model?_  Because I was wondering if there were significant improvement (decrease size or buildtime).  

I think that your answer is about building an application that use librealsense, Am i right?

NB: I am trying some options listed here to see if there is any significant improvments  : https://github.com/IntelRealSense/librealsense/wiki/Build-Configuration

MartyG-RealSense, 2020-10-27:
No, my answer was directed at your question of compilong for a specific cam model.

https://github.com/IntelRealSense/librealsense/issues/7587#issuecomment-709970978

I could find little information about how to do so via CMake, which is why the majority of the answer focused on how one might do model-recognition within the application code.  I did suggest though that if you needed the limitation to be applied during the build process only then you might be able to do so via some definitions in the **CMakeLists.txt** CMake build file.

[image]

closed 2020-11-11 by MartyG-RealSense