realsenseai/librealsense #7588: Frames are sometimes corrupted with a horizontal strip of an old frame

logidelic, 2020-10-16:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D414 | 
| Firmware Version                   | 5.12.7 | 
| Operating System & Version |   Ubuntu 18  | 
| Platform                                 | NVIDIA Jetson Xavier NX  |

### Issue Description
Usually things work well, but rarely, the camera gets into a state where every few frames are corrupted in a very particular way:

The top (~ 80%) of the frame is correct, but the bottom of the frame (~ 20%) is the bottom of an old frame.

It's the same old frame that is re-hashed, over and over for the bottom horizontal strip of the frame, but only every few frames. The frames in between these corrupted ones are fine. Restarting the camera fixes it.

Has anyone seen this? I've seen it on more than one machine/camera a handful of times.

MartyG-RealSense, 2020-10-16:
Hi @logidelic  There was a similar case in March 2020 on a Jetson Nano where the top 10% of the screen did not match up to the rest of the image, as though there was a second's difference between the timing of that top section and the rest.   Updating the SDK and reducing resolution helped in that case.

https://github.com/IntelRealSense/librealsense/issues/5986

logidelic, 2020-10-16:
Hi Marty,

Thank you so much for the response.

Unfortunately, even though it sounds coincidental, I believe this is a different issue:

- In my case, when I see these frames, the bottom part is always from the same old frame. That same, singular, old frame gets its bottom strip re-displayed with every one of the problematic frame pulls.

 - It's worth mentioning that, in my experimentation I've rebuilt librealsense in CPU mode as well as in CUDA mode. I've seen the problem with both builds of the library...

Thanks!

MartyG-RealSense, 2020-10-16:
Are you using **wait_for_frames()** in your project please, or are you using **poll_for_frames()** ?

wait_for_frames() blocks until a complete frame is available, whilst poll_for_frames() will not wait for a complete frame and returns instantly.  The differences between them are discussed in the link below. 

https://github.com/IntelRealSense/librealsense/issues/2422#issuecomment-423254709

logidelic, 2020-10-16:
Using `try_wait_for_frames()`.

MartyG-RealSense, 2020-10-16:
Okay, thank you very much.  In recent cases that I have seen **try_wait_for_frames()** used in, it may be masking problems because it prevents errors from being thrown and crashing the program (it returns a False state instead if there is a timeout) but you can instead have a situation where the program just appears to freeze up and do nothing.

logidelic, 2020-10-21:
Thanks again Marty. Switched to using `poll_for_frames` but eventually the problem reproed again. Some more info:

Something that I suspected and I think I can now confirm: It seems that when this problem occurs, it starts _**as soon as we enable the stream start streaming frames**_ . In other words, it doesn't suddenly manifest after the stream has been open for a while. I guess it is something in the stream init that (silently) goes wrong... I hope this is a good clue.

Attached is a video with an example of what I'm seeing when I have the problem. Note that (for this video) the top horizontal strip shows an old frame once every few frames.

Just to re-iterate: If I had some way of detecting this condition I'd be happy since it's so rare and I could just restart the stream in that case...

[corrupt.tar.gz](https://github.com/IntelRealSense/librealsense/files/5416059/corrupt.tar.gz)

MartyG-RealSense, 2020-10-21:
I watched the video intensely carefully.  There was no obvious source of the overlay outside of the video, even taking account for the possibility that it may be copy and pasting a section from lower down the image onto the top of it.

I focused on the image that appears in the window.  It looks very much like human hair.  What's more, it seems to be moving as the frames progress, as though the head is moving.  I see a human moving at the far edge of the video.

I wondered if the RealSense camera was being pointed at the window but a webcam such as one built into a laptop screen was pointing towards the back of the human's head, and somehow the computer webcam image was intermixing with the RealSense RGB image on the video.  Is this scenario possible?

logidelic, 2020-10-21:
HI Marty,

Thanks for watching my video. My covid-hair is now famous. To clarify about what you're seeing in that video:

- The bottom portion of the scene (the part that is not corrupt) does indeed show a human (me) on the left side. 
- The top strip overlay that periodically appears (every few frames) is static. I assure you that it is a portion of a single frame from the same sensor (possibly the first frame read). What's more it is in the correct position (it is not random or from another part of the frame; it is the top strip repeating itself where it originally showed up)
- There's no other camera.

MartyG-RealSense, 2020-10-21:
If auto-exposure is being used then the first frame or two tends to have incorrect exposure because the auto-exposure is settling down, so it is good practice to skip the first several frames at the start of streaming by adding code to a script to do so.  

`// Camera warmup - dropping several first frames to let auto-exposure stabilize`
`rs2::frameset frames;`
`for(int i = 0; i < 30; i++)`
`{`
`//Wait for all configured streams to produce a frame`
`frames = pipe.wait_for_frames();`
`}`

logidelic, 2020-10-21:
Hi Mary, was that comment meant for another thread?

MartyG-RealSense, 2020-10-21:
Oh yes, thanks very much for pointing it out quickly!  A case of too many browser tabs open at once.  :)

MartyG-RealSense, 2020-10-21:
I checked the comment and it was indeed meant for you.  I was wondering whether putting such a check into your application at its launch might eliminate any 'bad' frames at the very start of the streaming.

logidelic, 2020-10-21:
Heh, yea, I'll experiment with what you're suggesting, but as you point out, I don't think it'll fix the real part of the problem..

logidelic, 2020-10-21:
Just one more note: I haven't proved it yet, but I believe that this (or something similar) is sometimes happening with the depth stream as well. The symptom in that case is that the depth numbers don't make sense. As with the above, restarting the stream fixes the issue.

MartyG-RealSense, 2020-10-21:
If the problem only ever occurs at stream start then you may just need to reset once when the program starts.  Doing this in ROS is a good way to fix a range of errors at start-up, so it may be equally helpful in this case.

https://github.com/IntelRealSense/librealsense/issues/2161#issuecomment-408779150

If there are any log errors that are generated when the problem occurs, you could also set up a mechanism to catch **any** error (not just specific named ones) and do a reset if one occurs.  The error handling documentation of the SDK provides C++ code for error catching.

https://dev.intelrealsense.com/docs/librealsense-error-handling-scheme#section-types-of-errors

logidelic, 2020-10-22:
Regarding catching errors: We consider all exceptions to be fatal and restart everything in those cases, so I don't think this is the issue...

MartyG-RealSense, 2020-10-22:
I wonder if it would be a possibility to use D415's left infrared stream in **RGB8** color mode instead of using the RGB camera and see whether the flickering overlay still occurs.

[image]

closed 2020-11-07 by MartyG-RealSense