realsenseai/librealsense #7589: Unreal Engine pipeline start with config and a rs2::net_device results in start exception

elliotstronge, 2020-10-16:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 05.12.08.200 | 
| Operating System & Version |   Windows 10 / Raspberry Pi (Raspbian 10.6)   | 
| Kernel Version (Linux Only)    |  5.4.51                                       | 
| Platform                                 | PC/Raspberry Pi/ |
| SDK Version                            | 2.39                       | 
| Language                            |  UE4 c++                     | 
| Segment			|  Realtime Interactivity                       | 

### Starting the rs2 pipeline results in a device not connected error even though the network device exists

I'm using the realsense2 UE4 plugin to get depth frames from the camera. We need it to work over the network from a Raspberry Pi and it works without fail in the RealSense viewer, but fails when we try to start the pipeline in the RealsenseInspector in the plugin. The network device is properly enumerated and everything, we can pull the serial number but when it comes to starting the pipeline with the config, it fails and says the device is not connected.

I've double checked the proper net module is built and working with all source we're using, but it just doesn't seem to want to work unfortunately. Any help would be appreciated.

MartyG-RealSense, 2020-10-16:
Hi @elliotstronge  Is your network-connected D455 camera correctly identifying as a D455 in **BP_RealSenseInspector** above the serial number, please?

[image]

Also, can you confirm please that the serial number is appearing correctly in **BP_RealSenseInspector** in the Unreal wrapper on the central host computer please?  In other words, that it is retrieving the serial number through an ethernet cable that has a Pi with a D455 camera on the end of it.

elliotstronge, 2020-10-18:
Hi Marty,

It does when I add these lines to ARealsenseInspector::Start()
`rs2::net_device device("10.10.20.146");`
`rs2::context_ref RsContext(Context->GetHandle());`
`device.add_to(RsContext);`

This is what I get:

[image]

Which is the correct serial number according to Realsense Viewer. It's worth noting that when I play in editor from here, it seems to grab one frame and then time out with a bunch of messages that look like this: 

[image]

If I try the other method in URealsenseContext::QueryDevices(), i.e. this
`rs2::net_device device("10.10.20.146");`
`Devices.Add(NewDevice(device.get().get(), TEXT("RealSenseIPDevice")));`

It doesn't show up in the inspector, and also fails to find any connected devices even though the device exists in the URealsenseContext::Devices TArray.

elliotstronge, 2020-10-18:
Investigating further today has lead me to the root of the issue. I made a small C++ program to test different ways to enable the net device and what isn't working seems to be this line: 

`RsConfig.enable_device(std::string(TCHAR_TO_ANSI(*ActiveDevice->Serial)));`

in ARealSenseInspector::Start(); If I change that to: 

`rs2_context* ctx = Context->GetHandle();`
`rs2_device* dev = ActiveDevice->GetHandle();`
`rs2_context_add_software_device(ctx, dev, nullptr);`

I get frames coming through from the Raspberry Pi on the network again in the editor. It seems like a real hack, but good to find the issue and good to get something working.

MartyG-RealSense, 2020-10-19:
Thanks so much for sharing your workaround with the RealSense community @elliotstronge     :)

elliotstronge, 2020-10-19:
You're welcome. Worth adding to the bug tracker I reckon.

MartyG-RealSense, 2020-10-19:
I will refer your fix to Intel for consideration.  Thanks again!

MartyG-RealSense, 2020-10-20:
Hi again @elliotstronge   Would you consider contributing your workaround code as a **pull request** for merging into the RealSense SDK?  Doing so is completely optional though.  Thanks!

https://github.com/IntelRealSense/librealsense/pulls

elliotstronge, 2020-10-20:
I'd be happy to when I can find some time to wrangle one together that works well enough.

closed 2020-10-20 by elliotstronge

MartyG-RealSense, 2020-10-21:
@elliotstronge  Thanks very much!