realsenseai/librealsense #7590: How to read the start point by using SDK2.0 API

panpan0952, 2020-10-16:
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.04.01.02| 
| Operating System & Version |  win10  | 
| Platform                                 | PC |
| SDK Version                            | 2.36.0 | 
| Language                            | Python   | 
| Segment			|  Measurements  | 

### Issue Description
Hi, 
The Ch3.4 of  L515 datasheet Rev.002 noted that the zero point (ground zero reference) can be read via API.
I don't found this function in API. Can you give me some hints?

Two more questions:
Does every L515 have the same value(-4.5mm) of the ground zero offsets (the Z' in Fig.3.2 of L515 datasheet Rev.002)?
[image]
Does every L515 have the same value of the depth scale (0.00025000001187*****)?
or there were some slight differences in various L515s

RealSenseSupport, 2020-10-21:
Hi @panpan0952 

The values that you discuss here that is outlined in the datasheet is common amongst all L515 cameras.  Also the information should be available via intrinsic and extrinsic calls.

panpan0952, 2020-10-22:
Hi @RealSenseSupport 

Thanks for the reply, I checked the extrinsic and intrinsic parameters. I didn't found the offset distance between the front cover glass and the principle point. Or should I use a constant of 4.5mm as it? 

Here's the script I checked the  extrinsic and intrinsic parameters:

pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 1024, 768, rs.format.z16, 30)
config.enable_stream(rs.stream.infrared, 1024, 768, rs.format.y8, 30)
config.enable_stream(rs.stream.color, 1280, 720, rs.format.rgb8, 30)

align = rs.align(rs.stream.depth)
profile = pipeline.start(config)

frames = pipeline.wait_for_frames()
aligned_depth_frame = frames.get_depth_frame()
aligned_ir_frame = frames.get_infrared_frame()
aligned_rgb_frame = frames.get_color_frame()

extr = aligned_depth_frame.profile.get_extrinsics_to(aligned_rgb_frame.profile)
depth_intr = profile.get_stream(rs.stream.depth).as_video_stream_profile().get_intrinsics()
ir_intr = profile.get_stream(rs.stream.infrared).as_video_stream_profile().get_intrinsics()
rgb_intr = profile.get_stream(rs.stream.color).as_video_stream_profile().get_intrinsics()

pipeline.stop()

**Results:**
extr
Out[38]: 
rotation: [0.999972, -0.00588366, 0.00471838, 0.00601539, 0.999578, -0.0284068, -0.00454925, 0.0284344, 0.999585]
translation: [-0.000457802, 0.0139028, -0.0049672]

depth_intr
Out[39]: [ 1024x768  p[554.914 382.926]  f[735.484 735.82]  None [0 0 0 0 0] ]

ir_intr
Out[40]: [ 1024x768  p[554.914 382.926]  f[735.484 735.82]  None [0 0 0 0 0] ]

rgb_intr
Out[41]: [ 1280x720  p[636.257 354.875]  f[909.694 910.357]  Inverse Brown Conrady [0.175394 -0.544383 -0.000804691 0.000305413 0.495918] ]

RealSenseSupport, 2020-11-13:
Hi @panpan0952 

Use the constant outlined in the datasheet.  

One other measurement you may be interested in as well is the X-Y Depth Origin and this is taken from the mechanical image in the datasheet Figure 6-1 Intel(r) RealSense(tm) LiDAR Camera L515

Y - 26.15mm - From Bottom To Depth Origin
Depth origin location from X perspective is in the middle of the camera – Datum B

*If we don’t hear from you in 7 days, this issue will be closed.

panpan0952, 2020-11-16:
Hi @RealSenseSupport 
Thanks for your reply. This info is useful.

closed 2020-11-16 by panpan0952