realsenseai/librealsense #7591: Realsense startup failed

git-SteamedBun, 2020-10-16:
Realsense camera cannot start on Windows10. 
The realsense camera was recognized in the system, but failed in pipeline.start (), and no errors were reported. 
Why is that?

MartyG-RealSense, 2020-10-16:
Hi @git-SteamedBun  Could you check whether access to the camera is set to On in the camera section of the Windows privacy settings using the instructions in the link below please?  Thanks!

https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789

git-SteamedBun, 2020-10-16:
Yes, my camera permissions are on, and the same problem occurred when I used IntelRealsense-Viewer.exe. I couldn't open the interface. 
Is it because I don't have his operating environment on my computer?

MartyG-RealSense, 2020-10-16:
If you have problems launching the RealSense Viewer or it freezes after launch, could you update the graphics drivers on your computer please to make sure that the OpenGL graphics drivers are up to date (because the RealSense Viewer and the Depth Quality Tool programs rely on OpenGL).

git-SteamedBun, 2020-10-19:
After my graphics card driver update, realsense Viewer still won't start. I found that, realsense Viewer can start when I don't connect the camera, so I used VS to build the realsense environment to start the camera. I found that the function pipeline.start () would get stuck, causing the program to crash. How to solve this?

MartyG-RealSense, 2020-10-19:
If you have installed the full 802 mb RealSense SDK for Windows 10 from the SDK's **Releases** page, could you right-click on the RealSense Viewer launch shortcut icon on the Windows desktop and select the drop-down menu option **Open file location**.  

This will open the SDK's **Tools** folder.   Please double-leftclick on a file called **rs-capture** in the Tools folder to launch it.  If this rs-capture example runs successfully then it will suggest that the problem is related to the RealSense Viewer specifically.

git-SteamedBun, 2020-10-20:
I did the test in the way you said, but it failed. I think there may be something wrong with my win10 system. I'm going to change my computer or reinstall the system to test it.
Thank you very much for your reply.

MartyG-RealSense, 2020-10-20:
I look forward to your next test results.  Good luck!

git-SteamedBun, 2020-10-21:
https://github.com/IntelRealSense/librealsense/issues/6331#issuecomment-621650746

MartyG-RealSense, 2020-10-21:
I am not sure of the reason for linking to that comment.  Can you explain more please?  Thank you.

git-SteamedBun, 2020-10-21:
Yes, I want to know how to get the original IR image without reducing the access to depth information.

MartyG-RealSense, 2020-10-21:
Thank you for the information.  If you do not want to change the Laser Power setting or turn the IR emitter off then you can remove the IR pattern using the Visual Preset configuration called **Left Imager Color w/o IR Pattern** on the D415 model.  It will also be available for the D455 camera model in an upcoming firmware version.

https://dev.intelrealsense.com/docs/d400-series-visual-presets#section-preset-table

Using **Left Imager Color w/o IR Pattern** does not always work, and it did not work in a test with a D415 that I just performed.   In this situation, there is an alternative confiuration file to try:

https://github.com/IntelRealSense/librealsense/issues/6248#issuecomment-615902296

If that does not work either, my tests found that you can get back some depth detail lost from a turned-off IR emitter by manually reducing the **gain** value.  This causes the infrared image to become darker though, so it is a balancing act.

git-SteamedBun, 2020-10-21:
OK, thank you very much for your answer.

MartyG-RealSense, 2020-10-30:
Hi @git-SteamedBun   Do you still require assistance with this case, please?   Thanks!

git-SteamedBun, 2020-10-30:
No need for now, thank you very much, thank you

MartyG-RealSense, 2020-10-30:
Thanks very much for the fast response.   I will close this case now.   Thanks!

closed 2020-10-30 by MartyG-RealSense