realsenseai/librealsense #7592: 3D reconstruction -cloud of averaged frames from a .bag file

4nightmare, 2020-10-16:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   | 05.12.07.100 | 
| Operating System & Version |   {Win 10 | 
| Kernel Version (Linux Only)    |  -                                       | 
| Platform                                 | Win |
| SDK Version                            |  {2.38.1.2223 }                          | 
| Language                            |  python                          | 
| Segment			|  {others }                          | 

### Issue Description
Hi,
I just started coding some months ago and I'd like to create my own project with my RealSense D435i.
I'm trying to extract a point cloud generated from the averaged value obtained from more than one frame (like 10 ).

I saw that is possible to use rs-converter to exctract PLY of each frame, but then register each PLY will be really time consuming, so I was thinking to create an average point cloud from different frames and after this step save it as a PLY. Then I will probably register these averaged PLY file .

I don't need to create these averaged PLY in real-time so I was asking which way will be the best one to start, ( having some .bag file recorded).

Otherwise is there the possibility to use specific functionality of SDK-2 (or other packages) for having a "direct" 3D reconstruction from a .bag file? ( I saw that Open3D python package have a "reconstruction system", but it works using images recorded by their realsense_recorder.py and not with a .bag file or with the images extracted through rs-converter from a previous recording )

I saw also some software  like Dot3D that are able to create in "real-time" 3D reconstruction and I was wondering if someone developed a code for reaching the same objective from a .bag recorded file.

I hope to find some support here!

Bye

MartyG-RealSense, 2020-10-17:
Hi @4nightmare  If you took an average of more than one frame (e.g 10 frames) then there would be a risk of the result being skewed if one or more of the frames had a depth coordinate with a particularly large difference from the norm.  For example, for a 5 frame set, 1 + 3 + 1 + 1 + 2 / 5 would provide an average for that coordinate of 1.6 instead of 1).

I recall a case where advice was gven about using the RealSense MATLAB wrapper to convert a bag recorded with the RealSense SDK to a point cloud.   It was suggested that a MATLAB wrapper example program for point cloud generation could be edited so that it used a bag for the source of its camera data instead of the live camera.

https://github.com/IntelRealSense/librealsense/issues/3241#issuecomment-462496753

The RealSense wrapper for MATLAB is in the link below.  It requires Windows 10.  Fortunately, that is the OS that you are using.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/matlab

And the mentioned MATLAB point cloud example program is here:

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/matlab/pointcloud_example.m#L8

If you would prefer not to have to use an external tool such as MATLAB, the RealSense sample bag archive states that the same technique can be used to adapt SDK example programs to use bag files instead of real data.

https://github.com/IntelRealSense/librealsense/blob/master/doc/sample-data.md

**rs2::config cfg;
cfg.enable_device_from_file(<filename>);
pipe.start(cfg); // Load from file**

So whilst I have not tried this personally, conceivably you could adapt the C++ cfg code above for a Python example and use a bag file as the data source instead of live data.  An example of such a program in the Examples section of the RealSense Python wrapper is **opencv_pointcloud_viewer.py**

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_pointcloud_viewer.py

4nightmare, 2020-10-19:
Hi @MartyG-RealSense ,
Thank you for you fast and exhaustive reply.
I will try something following your suggestion!
I will update you ASAP.

In the meanwhile I'd like to ask you something else :
I updated SDK to the last stable version (v  2.38.1)  and consequentially also the Realsense-viewer.exe. After doing that I have problem in streaming .bag file recorded with previous version of SDK (and also previous version of camera's FW):
- I can't see RGB stream (only white screen with error "Hardware Timestamps unavailable" [I was able to play this file before the update]
- I had some crashes

I have tried to download a previous version of Realsense-viewer from Github and eith this the Rgb stream works, but when I try to create the pointcloud (in the "3D" viewer)  choosing the color stream as a texture it become balck and the depth stream disappear also in the "2D" part of the viewer.

I was also wondering if is it possible to apply a post processing filter threshold that is different from the one used in the recording phase? and if is it possible to do post-processing on recorded files that during the recording had the post-processing turned off?.

if needed i can open a new issue regarding these problems related to Realsense-viewwer

Let me know.
Thank you

MartyG-RealSense, 2020-10-19:
Could you download a bag from the **Sample Data** page and try dropping it into the center panel of the Viewer to see whether it plays successfully, please?  A bag should play irrespective of the currently installed SDK or firmware version.

https://github.com/IntelRealSense/librealsense/blob/master/doc/sample-data.md

As a test, I dropped the D435i pre-production bag from the Sample Data page into the Viewer of SDK 2.39.0 with firmware 5.12.8.200.   This bag would have been created in 2018, yet it played without any problems.

[image]

In regard to your question about whether you can adjust the Threshold filter during bag playback if a threshold has already been used during recording: I created a bag just now to test this and confirmed that changing the Threshold filter during playback will change the image irrespective of the Threshold that was set during recording.

A second bag recording test with all Post-Processing filters disabled confirmed that post-processing could be performed on playback.

4nightmare, 2020-10-20:
Hi,
Thanks for your feeedback!
Well I cannot play succesfully the video from sample-data  in the viewer.
I  attach a screenshot below:
As you can see the RGB stream doesn't work because of an "Hardware Timestamp unavailable" error
[image]

I don't understand why considering that I have updated SDK directly with the .exe file and I didn't modify anything in the installation path etc. it should works as before

Do you have some suggestions?... probably in these days i'll try to uninstall and reinstall it... you never know...

MartyG-RealSense, 2020-10-20:
I tested the walking bags from the sample data page on my laptop just now and had the same error message in the Color panel about per-frame metadata not being enabled when I tried to list the metadata.   As the bag metadata can be listed for the depth and IMU panels, it seems to be a problem related to the bag file contents rather than either of our computers.

I played Stairs.bag from the sample page in the 2.39.0 Viewer too and had the same issue with no data in the Color panel.  So I downloaded the 32.5 mb standalone 2.38.1 Viewer from the SDK **Releases** page's Assets list and all of the bags played correctly.  The 2.37.0 standalone Viewer also played the bags correctly.

I tested the standalone 2.39.0 Viewer and the bags played correctly.

Finally, I went back to the standard Viewer installed by the SDK (the one launched from the desktop) and the bag playback of color failed again.   So it seems to be an issue with the SDK-installed Viewer.

I was able to successfully view the sample bags in 3D point cloud mode in the standalone 2.39.0 Viewer.  I was also able to do so in the standalone 2.38.1 Viewer.

************

In conclusion, would it be possible to download the standalone Viewer for SDK **2.38.1** and see whether you still have problems please?

https://github.com/IntelRealSense/librealsense/releases/tag/v2.38.1

[image]

4nightmare, 2020-10-20:
Hi

I have uninstalled and re-installed SDK but the problems remained with the "desktop" viewer and I got the following error
[image]
[image]

Then following your suggestion I tested the standalone Viewer (for v2.37-v2.38-v2.39) and it always worked properly, but only after setting as a default processor my nvidia GPU; with integrated graphic the 3D view were not shown and the depth stream were stopped after trying to use the 3D view.

The only thing that I have noticed is the "warn" (below )in the terminal "Failed to convert"... but the viewer was working fine (I didn't check recording/ live streaming)
[image]

I will test in the next days recording and live streaming and in the meantime I will avoid the "desktop" viewer.

Thank you !

MartyG-RealSense, 2020-10-20:
Great news that you made progress!  Good luck with your tests. :)

MartyG-RealSense, 2020-10-30:
Hi @4nightmare   Do you have an update to provide about your tests please?   Thanks!

4nightmare, 2020-10-30:
Hi @MartyG-RealSense ,
regarding the problem with Intel Realsense Viewer, the standalone version works perfectly in playback, recording and creating the 3D pointcloud and mesh.
The viewer coming qith the SDK installation is still working bad, but this is not a problem for me to keep using the standalone version.
I hope that from the next update there will be a fix for that.

Regarding the Pointcloud reconstruction from a .bag file recorded, I'm still testing the matlab script following your suggestion and trying to adapting it to python.
At the same time I have found other interesting possible solutions like:

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv/kinfu  
http://introlab.github.io/rtabmap/
https://github.com/mp3guy/ElasticFusion
https://github.com/victorprad/InfiniTAM

and i'm checking if some of them will fit to me (it's a bit tricky for me to configure everything in the right way, so I'm a bit slow ).

So i think that's all for the moment for me... I will test a bit all of them.
Thanks for the support!

MartyG-RealSense, 2020-10-30:
Great feedback @4nightmare - good luck with your tests and please do share any new questions that arise from them!

shuixing85, 2020-11-05:
Hi All,
just wondering is there any process on this thread?

4nightmare, 2020-11-05:
Hi All,
Because of the covid and  my work situation I stopped my project for a while (I don't know for how long).
If you want close the thread no problem.
maybe I'll update you in future.
Thanks again for the support

MartyG-RealSense, 2020-11-05:
Thanks very much for the update @4nightmare - yes, let's close for now.   Good luck!

@shuixing85   If you have a similar problem, please feel free to create a new case about it using the **New Issue** button on the front page of this GitHub forum.   Thanks!

https://github.com/IntelRealSense/librealsense/issues/

closed 2020-11-05 by MartyG-RealSense