realsenseai/librealsense #7593: D435 cannot be recognized by realsense-viewer and python script

SUN-GE, 2020-10-16:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   |  RealSense Viewer 2.39.0 | 
| Operating System & Version |   Linux (Ubuntu 18.04)  | 
| Kernel Version (Linux Only)    |  4.15.0-121-generic                                        | 
| Platform                                 | PC  |
| SDK Version                            |   2.<?>.<?>                      | 
| Language                            |  python                         | 
| Segment			|  Robot                         | 

### Issue Description
I have installed the librealsense SDK and realsense-viewer. My D435 camera cannot be recognized by realsense-viewer. But my T265 camera could be recognized. And when I use 
$lsusb
Bus 002 Device 012: ID 8086:0b07 Intel Corp.
It shows the dev has connected to the system.

$ modinfo uvcvideo | grep ver
version:        1.1.2.realsense-1.3.14
description:    USB Video Class driver
srcversion:     BE1821F94C3D410077ACD10
vermagic:       4.15.0-121-generic SMP mod_unload 

$ modinfo uvcvideo | grep version
version:        1.1.2.realsense-1.3.14
srcversion:     BE1821F94C3D410077ACD10

SUN-GE, 2020-10-16:
I tried to connect the D435 camera by pyrealsense2:
import pyrealsense2 as rs
pipe = rs.pipeline()
profile = pipe.start()

and it return the error 
  File "test.py", line 3, in <module>
    profile = pipe.start()
RuntimeError: No device connected

But when I connect pc with my T265 camera, it works.

RealSenseSupport, 2020-10-20:
@SUN-GE How did you install librealsense SDK? By binary or build from source? Could you please check if https://github.com/IntelRealSense/librealsense/blob/master/config/99-realsense-libusb.rules is in the directory "/etc/udev/rules.d/"? If not, please copy to this directory to try realsense-viewer again. Thanks!

SUN-GE, 2020-10-20:
@RealSenseSupport  Thanks for your reply. I install the librealsense SDK from source. (follow this instruction: https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) . And the https://github.com/IntelRealSense/librealsense/blob/master/config/99-realsense-libusb.rules is already in the directory /etc/udev/rules.d/.

RealSenseSupport, 2020-10-21:
@SUN-GE Could you please check your BIOS if secure boot enabled? According to the user mentioned in https://github.com/IntelRealSense/librealsense/issues/5631#issuecomment-575115725 , he resolved such issue after disable secure boot. Please have a try. Thanks!

RealSenseSupport, 2020-10-29:
@SUN-GE Any update from your side? Looking forward to your update. Thanks!

RealSenseSupport, 2020-11-06:
@SUN-GE Could you please update? Please note that this issue will be closed if we don't get update from you for another 7 days. Thanks!

RealSenseSupport, 2020-11-12:
@SUN-GE Sorry that we didn't get update from you for week. Will close this issue now. Please feel free to create new one if you still have questions or issues. Thanks!

closed 2020-11-12 by RealSenseSupport