realsenseai/librealsense #7594: Depth consistently does not come up since flashing new firmware

pbeeson, 2020-10-16:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

 
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.12.08.200 | 
| Operating System & Version |   Ubuntu 18 | 
| Kernel Version (Linux Only)    |  (e.g. 5.4.0-48)                                         | 
| Platform                                 | Dell XPS Laptop  |
| SDK Version                            |  { 2.39.0-0~realsense0.3571 }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  {Robot }                          | 

### Issue Description

My development D415 was working fine for months.  Yesterday I flashed with new firmware after being prompted to by realsense-viewer.  Since then, my programs (and realsense-viewer) consistently come up with no depth stream, and the camera needs to be unplugged from USB and replugged in before it will work.

MartyG-RealSense, 2020-10-17:
Hi @pbeeson   Would it be possible please to downgrade your camera firmware to version **5.12.7.100** by visiting the firmware releases page and downloading the file **Signed_Image_UVC_5_12_7_100.bin** ?

After the download, if your camera is detected in the RealSense Viewer then you can go to the **More** option at the top of the Viewer's options side-panel and select the **Update Firmware** option from its drop-down menu.

[image]

Browse to the location where the *Signed_Image_UVC_5_12_7_100.bin**  file was downloaded to and select it.  The Viewer should now start installing that earlier firmware version int your camera.

Once the downgrade is completed, you have the option to try re-upgrading back to firmware version **5.12.8.200** by going back to the More option and this time selecting **Install Recommended Firmware** from its menu.  Doing so does not require you to download the firmware file for that version if you are using SDK 2.39.0, and the firmware update should begin automatically.

If you find that you continue to have problems with depth when using firmware 5.12.8.200 then you could downgrade back to 5.12.7.100 if that works for you.  Please bear in mind thouh that RealSense cameras have a limit of 20 downgrades, after which you can only upgrade to a newer firmware version, so please do not perform the downgrade repeatedly.

pbeeson, 2020-10-17:
Yeah, I’ll run downgraded for a few days next week and report back.

MartyG-RealSense, 2020-10-19:
Okay, thank you very much for the update.  I look forward to hearing how the camera behaves after a firmware downgrade.

akifh, 2020-10-20:
Same here and I can confirm that it works after downgrade to 5.12.7.100

pbeeson, 2020-10-20:
After downgrading, it seems to be behaving ok.  Given that @akifh also sees this, it sounds like the depth-based HDR of the newest firmware isn't very robust for the D415.

MartyG-RealSense, 2020-10-21:
@akifh   Thanks for your report!

@pbeeson  I will make a note of the problems that yourself and @akifh had with firmware 5.12.8.200 on D415 and inquire to Intel if I see similar cases.  Thanks!

MartyG-RealSense, 2020-10-30:
Hi @pbeeson    Are you continuing to get stable performance after your recent firmware downgrade, please?

You may also be interested in a new white-paper document Intel have publised about HDR if you have not seen it already.

https://github.com/IntelRealSense/librealsense/issues/7657

pbeeson, 2020-10-30:
I am.  The production firmware works great,  while the newer HDR firmware
definitely was affecting the depth image randomly on startup.  It was about
25 % of the time that it came up with *almost* all zero depth readings.

MartyG-RealSense, 2020-10-30:
@pbeeson   Thank you very much for the update.

MartyG-RealSense, 2020-11-07:
Hi @pbeeson   Do you require further assistance with this case, please?   Thanks!

pbeeson, 2020-11-11:
I don't believe so.  Hopefully the next release will fix this issue.

closed 2020-11-11 by pbeeson