realsenseai/librealsense #7595: D435i doesn't show up for any of the python wrapper examples or in the Realsense Viewer

17mccurdyt, 2020-10-16:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435i } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Linux (Ubuntu 19.04)| 
| Kernel Version (Linux Only)    |  4.9.140-tegra                                       | 
| Platform                                 | NVIDIA Jetson Xavier |
| SDK Version                            |  { 2 }                          | 
| Language                            |  {C#/python }                          | 
                     

### Issue Description
<Describe your issue / question / feature request / etc..>
D435i camera does not show up in the realsense viewer or for the provided examples.
The camera works on a windows machines and on the Nvidia Jetson it is being recognized by the usb port.

nvidia@nvidia-desktop-01:~$ rs-enumerate-devices 
No device detected. Is it plugged in?

nvidia@nvidia-desktop-01:~$ lsusb
Bus 002 Device 006: ID 8086:0b3a Intel Corp. 
Bus 002 Device 005: ID 0bda:0489 Realtek Semiconductor Corp. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 13d3:3549 IMC Networks 
Bus 001 Device 005: ID 046d:c016 Logitech, Inc. Optical Wheel Mouse
Bus 001 Device 004: ID 413c:2101 Dell Computer Corp. SmartCard Reader Keyboard
Bus 001 Device 002: ID 0bda:5489 Realtek Semiconductor Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
nvidia@nvidia-desktop-01:~$ 

nvidia@nvidia-desktop-01:~$ dmesg -wT
[Fri Oct 16 12:43:58 2020] Booting Linux on physical CPU 0x0
[Fri Oct 16 12:43:58 2020] Linux version 4.9.140-tegra (buildbrain@mobile-u64-2277) (gcc version 7.3.1 20180425 [linaro-7.3-2018.05 revision d29120a424ecfbc167ef90065c0eeb7f91977701] (Linaro GCC 7.3-2018.05) ) #1 SMP PREEMPT Thu Sep 24 16:09:59 PDT 2020
[Fri Oct 16 12:43:58 2020] Boot CPU: AArch64 Processor [4e0f0040]
[Fri Oct 16 12:43:58 2020] OF: fdt:memory scan node memory, reg size 48,
[Fri Oct 16 12:43:58 2020] OF: fdt: - 80000000 ,  2c000000
[Fri Oct 16 12:43:58 2020] OF: fdt: - ac200000 ,  44600000
[Fri Oct 16 12:43:58 2020] OF: fdt: - 100000000 ,  180000000
[Fri Oct 16 12:43:58 2020] earlycon: tegra_comb_uart0 at MMIO32 0x000000000c168000 (options '')
[Fri Oct 16 12:43:58 2020] bootconsole [tegra_comb_uart0] enabled
[Fri Oct 16 12:43:58 2020] Found tegra_fbmem: 00800000@a069c000
[Fri Oct 16 12:43:58 2020] Found lut_mem: 00002008@a0696000
[Fri Oct 16 12:43:58 2020] OF: fdt:Reserved memory: failed to reserve memory for node 'fb1_carveout': base 0x0000000000000000, size 0 MiB
[Fri Oct 16 12:43:58 2020] OF: fdt:Reserved memory: failed to reserve memory for node 'fb1_carveout': base 0x0000000000000000, size 0 MiB
[Fri Oct 16 12:43:58 2020] OF: fdt:Reserved memory: failed to reserve memory for node 'fb2_carveout': base 0x0000000000000000, size 0 MiB
[Fri Oct 16 12:43:58 2020] OF: fdt:Reserved memory: failed to reserve memory for node 'fb2_carveout': base 0x0000000000000000, size 0 MiB
[Fri Oct 16 12:43:58 2020] OF: fdt:Reserved memory: failed to reserve memory for node 'fb3_carveout': base 0x0000000000000000, size 0 MiB
[Fri Oct 16 12:43:58 2020] OF: fdt:Reserved memory: failed to reserve memory for node 'fb3_carveout': base 0x0000000000000000, size 0 MiB
[Fri Oct 16 12:43:58 2020] OF: reserved mem: initialized node ramoops_carveout, compatible id nvidia,ramoops
[Fri Oct 16 12:43:58 2020] OF: reserved mem: initialized node vpr-carveout, compatible id nvidia,vpr-carveout
[Fri Oct 16 12:43:58 2020] OF: reserved mem: initialized node grid-of-semaphores, compatible id nvidia,gosmem
[Fri Oct 16 12:43:58 2020] cma: Reserved 64 MiB at 0x00000000c2000000
[Fri Oct 16 12:43:58 2020] On node 0 totalpages: 2030589
[Fri Oct 16 12:43:58 2020]   DMA zone: 7152 pages used for memmap
[Fri Oct 16 12:43:58 2020]   DMA zone: 0 pages reserved
[Fri Oct 16 12:43:58 2020]   DMA zone: 457725 pages, LIFO batch:31
[Fri Oct 16 12:43:58 2020]   Normal zone: 24576 pages used for memmap
[Fri Oct 16 12:43:58 2020]   Normal zone: 1572864 pages, LIFO batch:31
[Fri Oct 16 12:43:58 2020] psci: probing for conduit method from DT.
[... 1258 lines cut ...]
[Fri Oct 16 12:50:48 2020] [<ffffff8008272a0c>] SyS_ioctl+0x8c/0xa8
[Fri Oct 16 12:50:48 2020] [<ffffff8008083900>] el0_svc_naked+0x34/0x38
[Fri Oct 16 12:50:51 2020] usb 2-3.3: usb_suspend_both: status 0
[Fri Oct 16 12:50:51 2020] usb 2-3: usb_suspend_both: status 0
[Fri Oct 16 12:50:51 2020] usb usb2: usb_suspend_both: status 0
[Fri Oct 16 12:51:23 2020] FAN rising trip_level:1 cur_temp:46100 trip_temps[2]:60000
[Fri Oct 16 12:52:53 2020] FAN cooling trip_level:0 cur_temp:37950 trip_temps[1]:46000
[Fri Oct 16 12:58:25 2020] FAN rising trip_level:1 cur_temp:46100 trip_temps[2]:60000
[Fri Oct 16 12:59:49 2020] FAN cooling trip_level:0 cur_temp:37950 trip_temps[1]:46000
[Fri Oct 16 13:05:01 2020] FAN rising trip_level:1 cur_temp:46100 trip_temps[2]:60000
[Fri Oct 16 13:06:25 2020] FAN cooling trip_level:0 cur_temp:37950 trip_temps[1]:46000
[Fri Oct 16 13:11:46 2020] FAN rising trip_level:1 cur_temp:46250 trip_temps[2]:60000
[Fri Oct 16 13:11:49 2020] usb 2-3.3: usb_suspend_both: status 0
[Fri Oct 16 13:11:49 2020] usb 2-3: usb_suspend_both: status 0
[Fri Oct 16 13:11:49 2020] usb usb2: usb_suspend_both: status 0
[Fri Oct 16 13:11:52 2020] usb 2-3.3: usb_suspend_both: status 0
[Fri Oct 16 13:11:52 2020] usb 2-3: usb_suspend_both: status 0
[Fri Oct 16 13:11:52 2020] usb usb2: usb_suspend_both: status 0
[Fri Oct 16 13:11:58 2020] usb 2-3.3: usb_suspend_both: status 0
[Fri Oct 16 13:11:58 2020] usb 2-3: usb_suspend_both: status 0
[Fri Oct 16 13:11:58 2020] usb usb2: usb_suspend_both: status 0
[Fri Oct 16 13:12:03 2020] usb 2-3.3: usb_suspend_both: status 0
[Fri Oct 16 13:12:03 2020] usb 2-3: usb_suspend_both: status 0
[Fri Oct 16 13:12:03 2020] usb usb2: usb_suspend_both: status 0
[Fri Oct 16 13:13:11 2020] FAN cooling trip_level:0 cur_temp:37800 trip_temps[1]:46000

MartyG-RealSense, 2020-10-17:
Hi @17mccurdyt   There is a similar case with D435i and Jetson (Nano in that case) that I am currently investigating.  Like in your case, the **RuntimeError: No device connected** error appears and the camera does not work with the Viewer or the Python wrapper.  

https://github.com/IntelRealSense/librealsense/issues/7480

I have added your case number to my notes so that I can follow up with you if there is a development on the other case.   Thank you very much.

MartyG-RealSense, 2020-10-25:
Hi @17mccurdyt   Thank you very much for your patience.  I do not have an Xavier board to conduct tests on.  I did though run tests with a Jetson Nano using a D435i and a D455 separately.  

In my own tests, both of the cameras were deteced by the Viewer and were able to stream depth, color and IMU successfully (aside from a pop-up warning every time when enabling the depth stream that did not prevent streaming from working correctly).  However, the cameras did not work with Python.  I have passed my test results on to Intel.

MartyG-RealSense, 2020-10-25:
During the past week there was a RealSense user with similar problems with the Python wrapper (again on a Jetson Nano) who succeeded when following a list of instructions that I provided for building librealsense and the Python wrapper from source.  I have shared the link to the instructions below so you can see whether they will work on your Xavier.

https://github.com/IntelRealSense/librealsense/issues/7480#issuecomment-713498309

17mccurdyt, 2020-10-26:
Hello, I've been following the comments on that thread and tried option 1 from the linked comment and am still running into the same issue where the camera is not being detected. I plugged it into my windows machine and updated the camera drivers however it is still having the same issue. I am going to try uninstalling and reinstalling one more time.

MartyG-RealSense, 2020-10-27:
Thanks for the report of your continued problems - please do update me with the results of your uninstall and reinstall when you have them.   Thanks!

17mccurdyt, 2020-10-29:
Hello again, I uninstalled everything and started with a fresh install
using the suggested method and everything worked, thanks!.

MartyG-RealSense, 2020-10-29:
Great to hear @17mccurdyt that you got it working - thanks very much for the update!

MartyG-RealSense, 2020-11-15:
Case closed due to no fuether comments received.

closed 2020-11-15 by MartyG-RealSense