realsenseai/librealsense #7599: using align method after getting frame throw socket

galikrosentalik, 2020-10-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400 | 
| Firmware Version                   | 05.11.11.100 | 
| Operating System & Version |   Linux Ubuntu 16  | 
| Platform                                 | PC |
| SDK Version                            |  2.35.0     | 
| Language                            |  C++    | 
| Segment			|  Robot   | 

### Issue Description
Hi everybody, 
my system architecture looks similar to this: 
`https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras.`
i have small cpu, intel UP `https://up-board.org/up`, connected to PC. 
My UP board is pretty loaded, he need to handle 3 Real-Sense devices, and few more tasks.
I noticed that the rs2::align.process(frameset) is taking too long, approximately 35 milliseconds, which causes my effective FPS to be lower then wanted 30 FPS.
when connecting the D435 directly to my PC for tests, the same rs2::align.process(frameset), is taking only 3 milliseconds... 

I am interested to spare unnecessary assignments, to re-locate the align.process to my PC instead of on the UP board.  
currently the UP board is sending the PC by socket RGB&depth frames.
rs::align API able to process only with frameset object.

Is there any way to construct frameset object from RGB&depth frames?? (of curse can supply intrinsics and extrinsics as well.. ).
Is there any way to send the frameset by socket?? (something similar to the get_data() method, to get frameset as byes.. ) 
Is there any way to use align by RGB&depth instead of frameset??
Thank's Gal.

dorodnic, 2020-10-19:
Hi @galikrosentalik 
The method we have to convert data back to frame objects is using software_device class:
https://github.com/IntelRealSense/librealsense/tree/master/examples/software-device
Another example using software device to convert PNGs to BAG in python: https://github.com/IntelRealSense/librealsense/blob/development/wrappers/tensorflow/tools/convert_to_bag.py
Align is expecting framesets. One way these can be generated is using Syncer class.

galikrosentalik, 2020-10-25:
Hi @dorodnic,
Thank you very much for your super fast comment :)
I used your example to refactor my code, works great...

closed 2020-10-25 by galikrosentalik