realsenseai/librealsense #7602: Is there any only playback-demo which is Compiled by Cmake？

1556260456, 2020-10-19:
I find it's hard to build the demo which is given in examples ,such as record-playback, because  all the demo are built in librealsense/build folder,there are top CMakelists.txt lower Cmakelists.txt,It‘s hard to understand the Library files and dependencies for newer. so,Is there any simple record-playback demo Compiled by Cmake? because I really need to read the bag files by c++,at the same time ,I also hope I can still use the API,such as filters and align.

MartyG-RealSense, 2020-10-19:
If you are new to librealsense and its libraries and dependencies, a good starting point for learning how to create your own applications before moving on to record and playback may be the **Getting Started** project.

https://github.com/zivsha/librealsense/tree/getting_started_example/examples/getting-started

I also looked into your difficulties with CMakeLists.txt regarding the **rs-record-playback** example.  The example only has a single CMakeLists.txt in its file list, so I was not certain of your meaning about top and lower files.  Could you provide more details about this, please?

https://github.com/IntelRealSense/librealsense/tree/master/examples/record-playback

1556260456, 2020-10-25:
this is the top level CMakeLists.txt of librealsense:
[image]
and here is the examples CMakeLists.txt which is the second level:
[image]
Finally this is the third CMakeLists.txt in the rs-record-playback:
[image]

And I notice this code in the top cmakelists.txt:
[image]
if It is mean that If build the unit tests mode,I can get an easier CMakeLists.txt in every demo code?
I really hope I can Compile each demo separately so that I can understand the details.

MartyG-RealSense, 2020-10-25:
Thank you for the clarification that your images provide about the CMakeLists.txt file locations.

If changes are made to an example and then the make command is used in the **build** folder then only the examples that have had changes made to them are built, not the entire SDK.   

https://github.com/IntelRealSense/librealsense/issues/2946

If your goal is to develop standalone versions of examples away from the folder structure of the SDK then the link below may be useful to you.

https://github.com/IntelRealSense/librealsense/issues/6797#issuecomment-657595834

1556260456, 2020-10-27:
Thanks for your reply.  I read the link you have listed ,And I still feel hard to Solve the problem. So ,let's simplify this problem:

whhen I copy the C++ code of rs-realsense-playback from examples to my own folder,and I write my own Cmakelists.txt like this:
[image]
[image]
After that ,I run _cmake .._  and _make_,I got the error like this,
[image]
Can you tell mehow to fix this error?Thanks very much

MartyG-RealSense, 2020-10-27:
Looking at the list of filenames at the bottom of the image above, it seems that the problem may be with the Visual Studio linkages to the **GLFW** graphics library.  

A RealSense community member has created a guide to setting up a Visual Studio record and playback project for RealSense with the three props files,

https://github.com/EduardoWang/visual-studio-real-sense-record-and-playback-project

Details about the location of the props files is here:

https://github.com/IntelRealSense/librealsense/issues/7106#issuecomment-678634741

1556260456, 2020-10-27:
I just use vscode as a editor on ubuntu16.04, and I Compile the C++ code with _cmake_ ,should I add some paths or dependencies about about GLFW?

MartyG-RealSense, 2020-10-27:
This subject is a little outside of my experience.  In the interests of not giving incorrect advice, I will refer the question to @dorodnic  the RealSense SDK Manager.

@dorodnic  Could you advise @1556260456 of the correct procedure for adding linkage to GLFW in a C++ project created in Ubuntu 16.04 and the vscode editor, please?   Thanks!

1556260456, 2020-10-27:
> I will refer the question to

thank you very much!

1556260456, 2020-10-28:
Because the problem of compiling demo separately cannot be solved in a short time，I also try to get date by ros. After I download  and install realsense-ros successfully,and I run **roslaunch realsense2_camera rs_camera.launch**,I got the error:
[image]
[image]
[image]
I was confused about it.

MartyG-RealSense, 2020-10-28:
If **control-transfer returned** errors only occur at roslaunch and then stop, or they occur at a rate of one one or two per minute after launch, then they can be safely ignored if the launch process has not exited.  You can check in **RViz** if the topics are being published correctly despite the messages.

If the **control_transfer returned** messages are generated continuously though then this can indicate a serious communication problem with the device.

1556260456, 2020-10-30:
before I see your advice,I reinstalled the SDK2.0 and update the FW,then the realsense-ros works well.But a new trouble appered:
when I open the realsense-viewer ,it shows that:
[image]
at the early time ,I can not record video ,But now It still shows this error,but can worlk normally.

MartyG-RealSense, 2020-10-30:
This is a rare error in my experience, though there was a reported case of the same error several weeks ago.  

https://github.com/IntelRealSense/librealsense/issues/7433#issuecomment-703236514

I am hesistant to suggest changes such as a firmware update if the Viewer and ROS are otherwise working normally, as it risks breaking a setup that is working aside from the annoyance of a pop-up box.

1556260456, 2020-10-30:
Thanks for your suggestion,If any update about:
[image]
Please let me know,  I will try  myself and also waiting for your good news!

MartyG-RealSense, 2020-11-07:
Reminder to @dorodnic to please reply to the request of @1556260456  above.  Thanks!

MartyG-RealSense, 2020-11-15:
Hi @ev-mp   Could you advise @1556260456 of the correct procedure for adding linkage to GLFW in a self-created standalone C++ project created in Ubuntu 16.04 and the vscode editor, please?   Thanks!

dorodnic, 2020-11-24:
https://github.com/IntelRealSense/librealsense/tree/master/examples/cmake example is showing how to link with **librealsense** using CMake. OpenGL and GLFW are 3rd-party components, not part of the SDK, so bundling them is outside the scope of this forum. Looking at the errors briefly, it seems the CMakeLists is missing OpenGL dependency. I'm not seeing `find_package(OpenGL REQUIRED)` in CMakeLists. For GLFW, download and build from source GLFW library according to their documentation, and then add glfw to target_link_libraries. Not sure we can help troubleshoot it further.

closed 2020-12-08 by MartyG-RealSense