realsenseai/librealsense #7603: using pyrealsense2 with L515

leemerlas, 2020-10-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Windows 10 | 
| Platform                                 | PC |
| SDK Version                            | 2.39.0.2337 | 
| Language                            |  Python | 
| Segment			|  {Robot/Smartphone/VR/AR/others } | 

### Issue Description

I just purchased the L515 Lidar camera. I want to use this camera with Python via the pyrealsense2 library and recently found the sample codes to be written in C++. I've read through the Python wrapper examples as well.

So my main question is, if I were to use pyrealsense2, is everything that's written in C++ (the examples) doable in Python as well? Are there any limitations?

Any help/advice is appreciated.

dorodnic, 2020-10-19:
Hi @leemerlas 
To the best of my knowledge 100% of functionality available in C/C++ is also available in python.

leemerlas, 2020-10-20:
Hi @dorodnic 

Thanks for the response! We want to implement our use case in Python, that's why I asked this question. Gonna leave this issue open for a bit more in case someone has more info to add.

RealSenseSupport, 2020-12-23:
@leemerlas 

Is there anything further needed here?

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2021-01-13 by RealSenseSupport