realsenseai/librealsense #7605: Python example unable to read from bag file

leemerlas, 2020-10-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Win 10 | 
| Platform                                 | PC |
| SDK Version                            |  2.39.0     | 
| Language                            |  Python 3.6.8 | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I was playing around with the Jupyter notebook example `depth_filters.ipynb` with the `stairs.bag` file and wanted to see if I could use this bag file with the other Python examples as well. After modifying `opencv_pointcloud_viewer.py` to read from the bag file, I ran the code in the terminal via `python opencv_pointcloud_viewer.py` but I was met with an error instead:

[image]

It seems it was unable to open the bag file. I tried running the same chunk of code but on Jupyter notebook instead, and it ran perfectly fine. My `pyrealsense2` version is `2.39.0`

Am I missing something here? Is this a bug/limitation?

dorodnic, 2020-10-19:
Hi @leemerlas 
Given the error, I can only suggest checking if the file exists at this path, accessible and valid. You can open it using the RealSense Viewer, before integrating into the sample.

leemerlas, 2020-10-20:
@dorodnic Hi, thanks for the response. 

I did open it using the RealSense Viewer first, and I can open it successfully. Haven't bothered to check if the file path is valid, should I use the absolute file path instead? I made sure to put the bag file in the same directory as `opencv_pointcloud_viewer.py` but maybe I need to check further.

RealSenseSupport, 2020-12-23:
Hi @leemerlas 

Were you able to check this further?  Are you still having issues?

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2021-01-02 by leemerlas