realsenseai/librealsense #7606: Cmake with V2.39.0 pyrealsense_net fatal error

tn2194, 2020-10-19:
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Win 10 | 
| Platform                                  | PC |

### Issue Description.
I build the pyrealsence from source .
Everything is fine until I build it by VS2017.
When it compiled the pyrealsense-net project it shows:
3>pyrs_net.obj : error LNK2001: unresolved external symbol rs2_get_failed_function
3>pyrs_net.obj : error LNK2001: unresolved external symbol rs2_get_error_message
3>pyrs_net.obj : error LNK2001: unresolved external symbol rs2_free_error
3>pyrs_net.obj : error LNK2001: unresolved external symbol rs2_get_librealsense_exception_type
3>pyrs_net.obj : error LNK2001: unresolved external symbol rs2_delete_device
3>pyrs_net.obj : error LNK2001: unresolved external symbol rs2_get_failed_args
3>C:\realsense\Release\pyrealsense2-net.cp37-win_amd64.pyd : fatal error LNK1120: 6 nresolved external symbol

It is important for me to use the pyrealsense-net package.I tried it with V2.38.1 and V2.39.0.
They are both failed.
Is any solution to solve it?
Thanks

dorodnic, 2020-10-19:
Hi @tn2194 
Please try building with -DBUILD_SHARED_LIBS=OFF to generate single DLL. I believe it should eliminate these errors.

tn2194, 2020-10-19:
@dorodnic Thank you very much. It finally works.
May I ask another question.

After  build it, I copy all DLL ,pyd and .lib to the site-package/pyrealsense2 folder.
When I **import pyrealsense2_net** it says ModuleNotFoundError: No module named 'pyrealsense2_net'.
Or **import pyrealsense2_net as rs**  rs.net_device('URL') it says AttributeError: module 'pyrealsense2' has no attribute 'net_device'.

How do I use correctly with pyrealsense2_net package?
Thanks.

RealSenseSupport, 2020-10-22:
@tn2194 Did you add cmake build flag -DBUILD_NETWORK_DEVICE=ON ?

adri1177, 2020-11-02:
Hello, i have the same problem and i am doing this from a script and the cmake is like this

`mkdir build && cd build
cmake ../ -DFORCE_LIBUVC=true -DCMAKE_BUILD_TYPE=release -DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS=bool:true -DPYTHON_EXECUTABLE=$(which python3) -DBUILD_NETWORK_DEVICE=ON
make -j8
sudo make uninstall && make clean && make && sudo make install`

What do i need to add so this works? @RealSenseSupport

RealSenseSupport, 2020-11-11:
@tn2194 @adri1177 rs-server needed for build network device according to https://github.com/IntelRealSense/librealsense/blob/master/tools/CMakeLists.txt#L21-L25 . However, rs-server supports Linux only according to
https://github.com/IntelRealSense/librealsense/blob/master/tools/rs-server/CMakeLists.txt#L24 

Could you please switch to Linux to try? Thanks!

RealSenseSupport, 2020-11-17:
@tn2194 @adri1177 Did you get chance to try with Linux? Any update from your side? Please note that this issue will be closed if we don't get your update in 7 days. Thanks!

adri1177, 2020-11-17:
@RealSenseSupport Hello, my test was made on a Jetson nano with Linux installed.
I am able to run rs-server and it is working fine, but I am not able to connect to that rs-server with python from another Linux computer.

RealSenseSupport, 2020-11-18:
@adri1177 Then I understood you have a different issue with this ticket. This ticket is for build issue in Win10 system. Did you get build issue under Linux? If it's not the same problem, could you please create another issue to clarify? Thanks!

RealSenseSupport, 2020-11-30:
@tn2194 Any other questions about this issue? Looking forward to your reply. Please note that the issue will be closed if we don't get response from your in 7 days. Thanks!

RealSenseSupport, 2020-12-08:
@tn2194 Sorry that we didn't hear from you for weeks. Closed the issue now. Please feel free to create another new one if you still have questions or issues. Thanks!

closed 2020-12-08 by RealSenseSupport