realsenseai/librealsense #7607: Using Depth and color frames as class attributes

noblegasses, 2020-10-19:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   | (5.12.06.00) | 
| Operating System & Version |  Linux (Ubuntu 16.04)  | 
| Kernel Version (Linux Only)    |  (5.4.0-51-Generic)                              | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.35.2 }                          | 
| Language                            |  {C++/OpenCV }                          | 
| Segment			|  {others }                          | 

### Issue Description
Hello, I have a quick Question that may be more of a result of me being new to c++, but I'm trying to use depth and color frames as class attributes for a realsense camera class. however, I get a no matching function to call error. My Header and Class are as follows:
### Header
```C++
#ifndef CAMERA_H`
#define CAMERA_H
#include <librealsense2/rs.hpp>
#include <opencv2/opencv.hpp>
#include <opencv2/highgui.hpp>
using namespace cv;
using namespace rs2;
class camera{
 private:
  rs2::config cfg;
  rs2::pipeline pipe;
  rs2::frameset frames;
  int ColorHeight;
  int ColorWidth;
  int DepthHeight;
  int DepthWidth;
  int fps;
  Mat color;
  Mat depth;
  depth_frame rsDepth;
  frame rsColor;
 public:
  //constructor to initialize camera
  camera(int ColorHeight, int ColorWidth, int DepthHeight, int DepthWidth, int fps);
  //a method to only return raw realsense frames, called in get_frames()
  void get_raw_frames();
  //gets camera data and converts them to opencv matricies
  void get_frames();
  //returns raw realsense frames
  depth_frame get_raw_depth();
  frame get_raw_color();
  //returns the opencv matricies, overwirtes input Mat
  void get_color(Mat &src);
  void get_depth(Mat &src);
  // displays the given opencv matrix
  void show_frame(Mat image);
};
#endif
```
### The Method Definitions
``` C++
#include <camera.hpp>
camera::camera(int ColorHeight, int ColorWidth, int DepthHeight, int DepthWidth, int fps){
 this->ColorHeight = ColorHeight;
 this->ColorWidth  = ColorWidth;
 this->DepthHeight = DepthHeight;
 this->DepthWidth  = DepthWidth;
 this->fps = fps;
 cfg.enable_stream(RS2_STREAM_DEPTH, DepthWidth, DepthHeight, RS2_FORMAT_Z16, fps);
 cfg.enable_stream(RS2_STREAM_COLOR, ColorWidth, ColorHeight, RS2_FORMAT_BGR8, fps);
 pipe.start(cfg);
}

 void camera::get_raw_frames(){
 frames = pipe.wait_for_frames();
 rsColor = frames.get_color_frame();
 rsDepth = frames.get_depth_frame();
}

void camera::get_frames(){
 this->get_raw_frames();
 depth = Mat(Size(DepthWidth, DepthHeight),CV_16U, (void*) rsDepth.get_data(), Mat::AUTO_STEP);
 color = Mat(Size(ColorWidth, ColorHeight),CV_8UC3, (void*) rsColor.get_data(), Mat::AUTO_STEP);
 depth.convertTo(depth, CV_8UC1, 255.0/1000);
}

void camera::show_frame(Mat image){
 int type = image.type();
 switch (type){
  case 0:
   namedWindow("Depth Image", WINDOW_AUTOSIZE);
   imshow("Depth Image", image);
   break;
 case 16:
  namedWindow("Color Image", WINDOW_AUTOSIZE);
  imshow("Color Image", image);
  break;}
}

void camera::get_color(Mat &src){
 this->color.copyTo(src);
}

void camera::get_depth(Mat &src){
 this->depth.copyTo(src);
}

depth_frame camera::get_raw_depth(){
  return this->rsDepth;
}

frame camera::get_raw_color(){
 return this->rsColor;
}
```
The error appears to be from the declaration of rsDepth in the header. How do I declare a depth frame for use within a method properly? Furthermore, can frames be returned from functions as a simple return, or do they have to overwrite a frame like I had to do for OpenCV matricies?

dorodnic, 2020-10-20:
Hi @noblegasses 
I don't see anything wrong with the code. What error are you getting? Are you building with C++11 support? Have you tried something like `depth_frame rsDepth = frame{};`? I suspect it may be related to missing default ctor in depth_frame class

noblegasses, 2020-10-24:
Hey, I'm sorry I didn't reply to this is sooner. 

I managed to solve the issue by changing rsDepth's type in the camera class to rs2::frame rather than depth_frame, but returning it in the main to a depth_frame. 

It appears that depth_frame cannot be initialized as blank, but can accept normal frames as input.

closed 2020-10-24 by noblegasses