realsenseai/librealsense #7608: D400 series stereo matching

sho-ji, 2020-10-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400                                                      | 

### Issue Description
When I shoot an object, the corners are rounded.
I think this phenomenon depends on the number of pixels in stereo matching.
How many pixels does the D400 series stereo matching use?

MartyG-RealSense, 2020-10-19:
Hi @sho-ji   The 400 Series cameras do not have a confidence threshold, so pixels are either 100% matched or they are not matched.  

https://github.com/IntelRealSense/librealsense/issues/3185#issuecomment-460540961

Aside from using the **Second Peak Threshold** setting as the above link suggests, you may also get less "blobbiness" in the image if you increase the value of the **Laser Power** setting.  

If you are applying a smoothing filter than that may also contribute towards rounded corners, as described in **2.  Edge-preserving filtering** of the section of Intel's post-processing guide linked to below.

https://dev.intelrealsense.com/docs/depth-post-processing#section-simple-post-processing

sho-ji, 2020-10-20:
Hi @MartyG-RealSense Thank you for your answer.
Stereo matching is not processed using only 1 pixel, isn't it?
If you don't use a few pixels, you won't be able to find a reliable depth.
And since it is rare that all the pixel values match, I think that the 1st peak will be acquired.
Therefore, I think there is a threshold for judging that they do not match.

Is there any material that describes that part in detail?

MartyG-RealSense, 2020-10-20:
The **Depth from Stereo** documentation provides more information about matching.

https://github.com/IntelRealSense/librealsense/blob/master/doc/depth-from-stereo.md

The **Basics of Stereo Depth** guide also provides information on matching, especially in the section headed **Steroscopic Vision**.

https://www.intelrealsense.com/stereo-depth-vision-basics/

sho-ji, 2020-10-20:
Thank you very much!

closed 2020-10-20 by sho-ji