realsenseai/librealsense #7611: How to run rs::align on GPU without CUDA

feefro, 2020-10-19:
I use d435 and d455 cameras.
OS linux
Processor ARM 64

I want aligned images but i do not have a cuda GPU, i need a fast alignment function. I need to use GPU.
I have seen that i can use rs2::gl::align 
https://github.com/IntelRealSense/librealsense/blob/058a172465285c1724b9f9075a63c8ac607197c3/include/librealsense2-gl/rs_processing_gl.hpp#L283

I cannot find a way to compile the library librealsense-gl.so without specifying CUDA option.
Is it possible? what is the compilation flag?
There is an example of rs2::gl::align?

MartyG-RealSense, 2020-10-19:
Hi @feefro  I hope that you find the links below useful.  You should not require CUDA in order to use GLSL Processing Blocks, as it is vendor-neutral (unlike CUDA, it does not require an Nvidia GPU specifically).  The improvements may not be noticable on low-end hardware though.

**rs-gl pros and cons and when to use it**
https://github.com/IntelRealSense/librealsense/pull/3654

**rs-gl example**
https://github.com/IntelRealSense/librealsense/tree/master/examples/gl

**rs-enabled alignment**
@bmegli did some work on GL-enabled align:

https://github.com/bmegli/realsense-network-hardware-video-encoder/issues/24

feefro, 2020-10-19:
With what flag is possible to compile librealsense2-gl.so? There is a building procedure to follow?

MartyG-RealSense, 2020-10-19:
I have not personally used GL but as far as I can tell, **librealsense2-gl.so** gets built at the same time that librealsense is built and so it is not something that you have to build yourself manually.  

If you are experiencing a conflict relating to librealsense2-gl.so then I recommend reading the link below.

https://github.com/IntelRealSense/librealsense/issues/4519

MartyG-RealSense, 2020-10-27:
Hi @feefro   Do you require further assistance with this case, please?   Thanks!

feefro, 2020-10-30:
no, thanks ;-)

MartyG-RealSense, 2020-10-30:
Okay, thanks very much for the update   :)

closed 2020-10-30 by MartyG-RealSense